platformio/platformio-core #893: Unable to build for feather_m0

eiginn, 2017-01-31:
Copy/paste from forum
A few days ago when recompiling a project from within the atom ide, it updated platforms. Since then I have been getting:

    /home/vaelen/Arduino/feather_m0_ble_test/src/feather_m0_ble_test.ino:31:17: error: 'A7' was not declared in this scope
    #define VBATPIN A7
    ^
from `platformio run` and within the ide.

platformio project https://github.com/eiginn/ble_animation_test
full run output https://gist.github.com/99dbfa171206a8807d7ead2cb7e5629d

    Platform Manager
    ================
    Platform atmelsam
    --------
    Updating atmelsam @ 1.6.0:      [Up-to-date]
    Updating framework-arduinosam @ 1.10611.0:      [Up-to-date]
    Updating toolchain-gccarmnoneeabi @ 1.40804.0:  [Up-to-date]

Not sure what would have changed to do this

### Configuration

Linux (Ubuntu 16.04)
PlatformIO, version 3.2.1

#### Steps to Reproduce

1. `git clone https://github.com/eiginn/ble_animation_test.git && cd ble_animation_test`
2. `platformio run -v`

eiginn, 2017-01-31:
was suggested by to downgrade atmelsam platform and this did work via:

```
[env:adafruit_feather_m0_usb]
platform = atmelsam@1.5.0
```

ivankravets, 2017-01-31:
Please move to this repo https://github.com/platformio/platform-atmelsam/issues

Sorry.

closed 2017-01-31 by ivankravets