platformio/platformio-core #894: Setting project-option fails for Travis build

Lauszus, 2017-02-01:
The following commit fails: https://github.com/felis/USB_Host_Shield_2.0/commit/7b0487da80d71729a9dbe92100c23eb9915e094e.

Here is the Travis build: https://travis-ci.org/felis/USB_Host_Shield_2.0/builds/197340237.

ivankravets, 2017-03-26:
Please sorry for the late answer. I've just forked your repo and tried to play with it. I can't reproduce this issue. See

* https://github.com/ivankravets/USB_Host_Shield_2.0/commit/72808e42c1ce12fad0e5c87df8eed7587a75cf6f
* https://travis-ci.org/ivankravets/USB_Host_Shield_2.0/jobs/215278224#L245
* https://travis-ci.org/ivankravets/USB_Host_Shield_2.0/jobs/215278224#L291

Lauszus, 2017-03-26:
The thing is that I only wanted to disable that warning for the ESP8266, thus the reason I wanted to use ```--project-option="build_flags=-Wno-unused-function"```, but that did not work.

Instead I had to modify ```PLATFORMIO_BUILD_FLAGS``` as shown in: https://github.com/felis/USB_Host_Shield_2.0/commit/14cf737d7d4c1ca77e85a8946ad147a6da5457e4.

ivankravets, 2017-03-26:
Ahh... Yes, that was a bug. I fixed it 1 month ago in https://github.com/platformio/platformio-core/commit/2467d5a5d097d583fffeef78eb2e027d50a745d2

PIO Core 3.3 is planned for releases tomorrow. ~4 months since the previous release 😊

closed 2017-03-27 by ivankravets

Lauszus, 2017-03-28:
Great!