platformio/platformio-core #895: Make PlatformIO Python 3 compatible

StefanD986, 2017-02-03:
What kind of issue is this?

- [x] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

------------------------------------------------------------------

### Configuration

**Operating system**:
Windows and Mac

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 3.2.1

### Description of problem
PlatformIO is not compatible to Python 3 (as stated on the website). It would be nice and would ease the install process if PlatformIO would also work with Python 3.

ivankravets, 2017-02-03:
PlatformIO's Build System depends on [SCons](http://scons.org) tool that doesn't support Python 3. Please note that SCons team works in this way.

skorokithakis, 2017-08-05:
@ivankravets How hard would it be to remove this dependency, or replace it with something else?

Would it be feasible to write an adapter module that called out to SCons from Python 3? Although that would be a hassle when installing...

Python 2 is quickly becoming irrelevant, and I'm finding myself having to keep a Python 2 env around just for PlatformIO. I think this should be a higher priority, as PlatformIO risks finding itself in a position where it can't be easily installed soon (with OSes moving to Py3).

I would be willing to contribute some code to the migration to Py3.

ivankravets, 2017-08-05:
>  PlatformIO risks finding itself in a position where it can't be easily installed soon (with OSes moving to Py3).

Python 2 is a stable and supported branch. I think, minimum 5 years all major OSs will include Python 2 distribution by default (as they now do). Hundred OS-dependent libraries and software are written in Python 2: macOS, Linux, BSD.

Nevertheless, SCons Team [hardly working on the first Python3 compatible release](https://bitbucket.org/scons/scons/commits/all).

- https://bitbucket.org/scons/scons/src/9ef88cce4c52e65083beba3f416d45a494855693/src/CHANGES.txt?at=default&fileviewer=file-view-default

tehnorm, 2017-09-28:
Just noticed that it looks like SCons 3.0.0 just released support for python 3.5+!

http://scons.org/scons-300-is-available.html

ivankravets, 2018-01-26:
Upgraded to SCons 3.0 in PlatformIo Core 3.5.2 ( https://github.com/platformio/platformio-core/issues/1253 ). Now, need to upgrade PIO Core to Python 3.

mintisan, 2018-04-29:
How can I install the specific version for Python3?

It seems to be not OK, if I use `pip3 install platformio`.

```
➜  ~ pip3 install platformio
Collecting platformio
  Downloading https://files.pythonhosted.org/packages/7c/95/899efc1264c63d1ba1d32b4160b2fe3d48020de00f7b8c91dc5b712aa05f/platformio-3.5.2.tar.gz (95kB)
    100% |████████████████████████████████| 102kB 238kB/s
    Complete output from command python setup.py egg_info:
    PlatformIO Core v3.5.2 does not run under Python version 3.5.1 (default, Jul  9 2016, 11:42:58)
    [GCC 4.2.1 Compatible Apple LLVM 7.0.2 (clang-700.1.81)].
    Minimum supported version is 2.7, please upgrade Python.
    Python 3 is not yet supported.

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /private/var/folders/7z/j_bymh4x3l7cp_m_xgk132mr0000gn/T/pip-install-dnnmbov9/platformio/
```

Peque, 2018-04-30:
@mintisan As you can see in your `pip3` command output: "Python 3 is not yet supported".

That means Platformio is currently not Python 3 compatible. Hence this issue. :wink:

You may want to try `pip2 install platformio` (within a system/virtual environment that uses Python 2 instead).

Navis-Raven, 2018-08-21:
Why platformio is running on python 2.7 and not on python 3

ivankravets, 2018-12-26:
Hi all,

Sorry for the delay. We have finally added full support for Python 3.5+. Of course, Python 2.7 is also in the supported list. 

We would be thankful if you re-test with `$ pio upgrade --dev`. It should be PlatformIO 4.0.

Does it work for you?

closed 2018-12-26 by ivankravets

13rac1, 2018-12-26:
Awesome! Confirming install/run in Ubuntu 18.04 LTS! :tada: :+1: An existing platform.io project builds without issue. The tests nearly pass; failures could be an environment issue.
```
$ python --version
Python 3.6.7
$ python3 -m venv venv # snip
$ source venv/bin/activate # snip
$ python setup.py install # snip
$ platformio --version
PlatformIO, version 4.0.0a1
$ pip install pytest # snip
$ make test 
py.test -v -s -n 3 --dist=loadscope tests --ignore tests/test_examples.py --ignore tests/test_pkgmanifest.py
usage: py.test [options] [file_or_dir] [file_or_dir] [...]
py.test: error: unrecognized arguments: -n --dist=loadscope tests
  inifile: None
  rootdir: /home/user/platformio-core
Makefile:13: recipe for target 'test' failed
make: *** [test] Error 4
# ¯\_(ツ)_/¯
$ py.test -v -s --ignore tests/test_examples.py --ignore tests/test_pkgmanifest.py
# Nearly passes! Only three failures, the rest is snipped:
tests/commands/test_ci.py::test_ci_boards FAILED
tests/commands/test_ci.py::test_ci_project_conf FAILED
tests/commands/test_test.py::test_local_env FAILED
======================================================= FAILURES =======================================================
____________________________________________________ test_ci_boards ____________________________________________________

clirunner = <click.testing.CliRunner object at 0x7f69a40f83c8>
validate_cliresult = <function validate_cliresult.<locals>.decorator at 0x7f69a6a4e400>

    def test_ci_boards(clirunner, validate_cliresult):
        result = clirunner.invoke(cmd_ci, [
            join("examples", "wiring-blink", "src", "main.cpp"), "-b", "uno", "-b",
            "leonardo"
        ])
>       validate_cliresult(result)

tests/commands/test_ci.py:32: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

result = <Result SystemExit(2,)>

    def decorator(result):
>       assert result.exit_code == 0, result.output
E       AssertionError: Usage: ci [OPTIONS] [SRC]...
E         
E         Error: Invalid value for "src": Found invalid path: examples/wiring-blink/src/main.cpp
E         
E       assert 2 == 0
E        +  where 2 = <Result SystemExit(2,)>.exit_code

tests/conftest.py:27: AssertionError
_________________________________________________ test_ci_project_conf _________________________________________________

clirunner = <click.testing.CliRunner object at 0x7f69a40f83c8>
validate_cliresult = <function validate_cliresult.<locals>.decorator at 0x7f69a6a4e400>

    def test_ci_project_conf(clirunner, validate_cliresult):
        project_dir = join("examples", "wiring-blink")
        result = clirunner.invoke(cmd_ci, [
            join(project_dir, "src", "main.cpp"), "--project-conf",
            join(project_dir, "platformio.ini")
        ])
>       validate_cliresult(result)

tests/commands/test_ci.py:41: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

result = <Result SystemExit(2,)>

    def decorator(result):
>       assert result.exit_code == 0, result.output
E       AssertionError: Usage: ci [OPTIONS] [SRC]...
E         
E         Error: Invalid value for "-C" / "--project-conf": Path "examples/wiring-blink/platformio.ini" does not exist.
E         
E       assert 2 == 0
E        +  where 2 = <Result SystemExit(2,)>.exit_code

tests/conftest.py:27: AssertionError
____________________________________________________ test_local_env ____________________________________________________

    def test_local_env():
        result = util.exec_command([
            "platformio", "test", "-d",
            join("examples", "unit-testing", "calculator"), "-e", "native"
        ])
        if result['returncode'] != 1:
>           pytest.fail(result)
E           Failed: <unprintable Failed object>

tests/commands/test_test.py:28: Failed
```

rab852, 2018-12-29:
Using a windows machine I can download the dev branch but still has the python 2.7.13 dependency

ivankravets, 2019-01-01:
@rab852 PlatformIO IDE currently depends on Python 2.7. Please open an issue here https://github.com/platformio/platformio-vscode-ide/issues

AlexFBP, 2019-02-09:
Just saying...

```
$ pip install -U platformio
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020.
Please upgrade your Python as Python 2.7 won't be maintained after that date.
A future version of pip will drop support for Python 2.7.
```

ivankravets, 2019-02-09:
You can use PlatformIO Core 4.0 which supports Python 3. See https://docs.platformio.org/en/latest/installation.html#development-version

Navis-Raven, 2019-02-15:
@ivankravets  When PlatformeIO Core 4.0 will be released ?

ivankravets, 2019-02-15:
See https://github.com/platformio/platformio-core/milestone/57

commit a60c57ac (2018-12-26): Initial support for Python 3.5+ // Resolve #895 Resolve #1365