platformio/platformio-core #897: CLion code insights doesn't work inside ./lib folder

diejmon, 2017-02-05:
What kind of issue is this?

- [x] PlatformIO Core. If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: macOS 10.12.3

**PlatformIO Version** (`platformio --version`): PlatformIO, version 3.3.0a10

### Description of problem
CLion 2016.3.2 code insights doesn't work in `./lib` folder unless all files inside this folder are added to `CMakeLists.txt`. For example: `add_executable(${PROJECT_NAME} ${SRC_LIST} lib/DHT/DHT.h lib/DHT/DHT.cpp)`. But `CMakeLists.txt` is regenerated during `PLATFORMIO_REBUILD_PROJECT_INDEX` build step.  

#### Steps to Reproduce

1. `platformio init --ide clion --board uno`
2. Import project into CLion
3. Create a folder inside `./lib` folder
4. Create files inside the folder

### Actual Results
All files inside `./lib` folder appears in light grey inside CLion project navigator and code insights doesn't work.

### Expected Results
CLion recognize all files inside `./lib` folder similar to how it recognize them inside `./src` folder.

diejmon, 2017-02-06:
I've found a workaround which is acceptable for me: Right click on `./lib` folder inside CLion and choose `Mark Directory as -> Library Files`. After this code insights are working inside `./lib`.

ivankravets, 2018-02-13:
Please re-test with the latest PIO Core and re-init project:
```
pio upgrade --dev
pio init
```

commit 0c2f9734 (2018-02-13): Mark project source and library directories for CLion IDE // Resolve #1359 Resolve #897 Resolve #1345
closed 2018-02-13 by ivankravets