platformio/platformio-core #898: Always recieve ISO C99 warning at build

SRGDamia1, 2017-02-06:
### Configuration

**Operating system**:  Windows 7 or TravisCI

**PlatformIO Version** (`platformio --version`): 3.2.1

### Description of problem
At every build, I receive this error:
 <command-line>:0:17: warning: ISO C99 requires whitespace after the macro name

The GCC linter in the IDE also displays this warning for every project.

The build continues successfully, but the warning is always there.

I also see the error in builds run by TravisCI.  For example:  https://travis-ci.org/EnviroDIY/EnviroDIY_Mayfly_Logger/jobs/198466734

ivankravets, 2017-02-06:
Please move issue to related repository. Thanks!

* https://github.com/platformio/platform-atmelavr/issues

closed 2017-02-06 by ivankravets

SRGDamia1, 2017-02-06:
Is it an AtmelAVR specific problem?  It seems like a compiler problem to me because even TravisCI sees the issue and it's not writing to any hardware.

ivankravets, 2017-02-06:
Yep, see `platform = atmelavr`. This repository is intended for PlatformIO Core issues.

SRGDamia1, 2017-02-06:
Moved to https://github.com/platformio/platform-atmelavr/issues/29