platformio/platformio-core #901: VCS revision "on-the-fly" example doesn't work

ChrisAlphabet, 2017-02-15:
Hi all,

I am trying to use the build flag example that sets a build flag for VCS revision

```ini
[env:exec_command]
; get VCS revision "on-the-fly"
build_flags = !echo "-DPIO_SRC_REV="$(git rev-parse HEAD)
```

but I get an error that says

*** [.pioenvs\nodemcu\firmware.elf] Implicit dependencyC:\Users\ckolo.platformio\platforms\e
spressif8266\builder\rev-parse' not found, needed by target .pioenvs\nodemcu\firmware.elf'.

To get around this I used

```
git describe --abbrev=4 --dirty --always --tags > temp
SET /P VERSION= < temp
del temp
SET PLATFORMIO_BUILD_FLAGS='-DGIT_VERSION="%VERSION%"'
```

but this overwrites the build flags in the `.ini` file (see [here](https://community.platformio.org/t/pass-global-define-at-run-time/1439/6?u=chrisalphabet))

I think the first issue is a bug and would like to know if the second issue is a bug or expected behaviour.

Thanks

ivankravets, 2017-02-18:
> but this overwrites the build flags in the .ini file

Fixed in https://github.com/platformio/platformio-core/commit/2467d5a5d097d583fffeef78eb2e027d50a745d2

ivankravets, 2017-02-18:
This example is for Unix. You can run any command using `!` prefix.

danielcrowley, 2017-03-06:
Hi Ivan, 
Is there an example of how this would run on Windows?

Regards,
Daniel

ivankravets, 2017-03-06:
See updated examples 
* http://docs.platformio.org/en/latest/projectconf.html#build-flags

closed 2017-03-06 by ivankravets

ivankravets, 2017-03-06:
Dynamic Build Flags!

* http://docs.platformio.org/en/latest/projectconf.html#dynamic-build-flags

danielcrowley, 2017-03-07:
Excellent, thanks Ivan!

commit a235c532 (2017-03-06): Add example how to get VCS revision "on-the-fly" for the Windows // Resolve #901