platformio/platformio-core #903: *cpp/*.h from "lib" folder not found

SRGDamia1, 2017-02-16:
What kind of issue is this?

- [ x] PlatformIO Core. If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 7, Atom 1.14.2 ia32, PIO IDE 2.0.0-alpha.5

**PlatformIO Version** (`platformio --version`): version 3.3.0a10

### Description of problem
Compiler does not find files in the "lib" directory

#### Steps to Reproduce
1. Add *.cpp and *.h to project ./lib
2. Call *.h from *.ino

### Actual Results
Compilation terminated "No such file or directory"

### Expected Results
Successful compilation

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini

[platformio]
src_dir = examples/modular_sensors/
lib_dir = libs

[env:envirodiy_mayfly]
platform = atmelavr
board = mayfly
framework = arduino
lib_ldf_mode = deep+
lib_deps =
    https://github.com/EnviroDIY/Libraries.git
```

**Source file to reproduce issue:**
```cpp
#include "Sensor.h"
#include "MayFlyOnboardTempSensor.h"
#include <MayFlyExampleSensor1.h>
```
[image]

SRGDamia1, 2017-02-16:
I was able to get it to work by invoking "build_flags = -I./lib" but I didn't think that should be necessary.

ivankravets, 2017-02-18:
Please take a look how libraries should be organized:

* https://github.com/platformio/platformio-examples/tree/develop/atmelavr-and-arduino/arduino-external-libs/lib

closed 2017-02-18 by ivankravets