platformio/platformio-core #904: PlatformIO doesn't upload firmware correctly to Arduino Leonardo

The00, 2017-02-17:
### Configuration

**Operating system**: Windows 7 64bit

**PlatformIO Version** (`platformio --version`): 3.2.1

### Description of problem
Hi !
When uploading code to an arduino leonardo, the normal serial port doesn't re-appears after uploading. Instead, the arduino stays on the bootloader serial port. 

#### Steps to Reproduce

1. create project for Eclipse IDE: `platformio init --ide eclipse --board leonardo`
2. Import project in Eclipse and add main.cpp source code
3. Upload a first time. It should work, but the serial port is lost. You will see instead  "Arduino leonardo bootloader" in the device manager.
4. Re-upload. It will fail.
5. You can unbrick the board by uploading a sketch using Arduino IDE(1.8.1). (set the port corresponding to Arduino leonardo bootloader). The normal serial port re-appears (Arduino Leonardo). 

### Actual Results
After an upload, the serial COM present is the bootloader, not the normal COM port. The uploaded code is running (blink demo code, tested with several delay). But the Serial communication doesn't work, and it's not possible to re-upload using Platformio.

### Expected Results
For a normal uploading, the IDE reset the board. The bootloader shows up, the code is uploaded, and then the normal serial port goes on. Here, it doesn't, even after manual reset.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:leonardo]
platform = atmelavr
board = leonardo
framework = arduino
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>
#define WLED	13  // Most Arduino boards already have an LED attached to pin 13 on the board itself

void setup()
{
  pinMode(WLED, OUTPUT);  // set pin as output
  Serial.begin(9600);
}

void loop()
{
  digitalWrite(WLED, HIGH);
  delay(200);
  digitalWrite(WLED, LOW);
  delay(200);
  Serial.println("test");
}
```

### Additional info
 This problem looks like #178 , but according to the thread, the issue should have been solved in the 1.5.0 version. I tried with two Leonardo, same result on both.

The00, 2017-02-17:
I tried on Linux, and it's work. So it may be a compatibility problem with the windows driver of the board (installed by the arduino IDE).

ivankravets, 2017-02-18:
Please move issue to https://github.com/platformio/platform-atmelavr/issues

closed 2017-02-18 by ivankravets