platformio/platformio-core #908: installing library passing a tag or branch fails both from cmdline and from the ide

dpeddi, 2017-02-27:
### Configuration

**Operating system**:
Win7_64

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 3.2.1

### Description of problem
installing library passing a tag or branch fails both from cmdline and from the ide

platformio lib install https://gthub.com/adafruit/DHT-sensor-library.git#1.1.1
Library Storage: D:\Apps\Atom x64\projects\souliss\.piolibdeps
LibraryManager: Installing DHT-sensor-library
git version 2.11.1.windows.1
Cloning into 'D:\Apps\Atom x64\projects\souliss\.piolibdeps\installing-xl0zvj-p
ckage'...
fatal: unable to update url base from redirection:
  asked for: https://github.com/adafruit/DHT-sensor-library.git#1.1.1/info/refs
service=git-upload-pack
   redirect: https://github.com/adafruit/DHT-sensor-library

#### Steps to Reproduce

1. platformio lib install https://gthub.com/adafruit/DHT-sensor-library.git#1.1.1
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

ivankravets, 2017-02-27:
PlatformIO IDE > Home > Switch to PIO IDE 2.0 Developer Preview

Does it work now?

ivankravets, 2017-02-27:
> platformio lib install https://gthub.com/adafruit/DHT-sensor-library.git#1.1.1

You have a mistake in GitHub domain.

closed 2017-02-27 by ivankravets

dpeddi, 2017-02-27:
It works, thanks...