platformio/platformio-core #909: can't build

dpeddi, 2017-02-27:
### Configuration

**Operating system**:
Win7_64

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 3.3.0a12

### Description of problem
Can't build a project using both Arduino-Time and  function included in #include <time.h>

see https://github.com/PaulStoffregen/Time/pull/67 as well...

#### Steps to Reproduce

add following lib to platformio.ini
lib_deps  =  
        https://github.com/PaulStoffregen/Time.git#v1.5
        ESPAsyncTCP=https://github.com/me-no-dev/ESPAsyncTCP.git
        ESPAsyncWebServer=https://github.com/me-no-dev/ESPAsyncWebServer.git

try to compile ESPAsyncWebServer example and add to its include #include <TimeLib.h>

### Actual Results
piolibdeps\ESPAsyncWebServer\src\WebHandlers.cpp: In member function 'AsyncStaticWebHandler& AsyncStaticWebHandler::setLastModified(time_t)':
.piolibdeps\ESPAsyncWebServer\src\WebHandlers.cpp:72:60: error: 'gmtime' was not declared in this scope
return setLastModified((struct tm *)gmtime(&last_modified));

### Expected Results
If you build under ArduinoIDE it build correctly

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

when ArduinoIde builds library to include into the project (in this case Arduino-time) it produce time.c.o, while platformio produce time.o so it conflicts with gcc internals..

I tried to ask to library developers to rename its time.c and time.h into time_<somethingelse> but he refuse to do so :-( and suggests to ask you for support

Regards

valeros, 2017-03-01:
Hi @dpeddi, sorry for the late reply!
As a workaround, could you please try to rename `Time.h` to `_Time.h` and build your project again?

dpeddi, 2017-03-01:
As you can see on pull request  https://github.com/PaulStoffregen/Time/pull/67 I already found a workaroud... but the original developer refuse to accept it.

He says that since on Arduino Ide it works, this is your problem...

ivankravets, 2018-06-18:
> roduce time.c.o, while platformio produce time.o so it conflicts with gcc internals..

We added "EXT.O" to Arduino framework in the latest releases. Should work now.

ldab, 2019-11-22:
> 
> 
> > roduce time.c.o, while platformio produce time.o so it conflicts with gcc internals..
> 
> We added "EXT.O" to Arduino framework in the latest releases. Should work now.

Is this fixed? I'm experiencing the same issue, as referenced here https://community.platformio.org/t/esp8266-ssl-compile-error/4582/7

valeros, 2019-11-22:
Hi @ldab ! The problem is related to the header file `Time.h` and since you're using case insensitive OS the only way is to rename this file.

ivankravets, 2022-04-02:
See solution in https://github.com/platformio/platformio-core/issues/909#issuecomment-557470263

closed 2022-04-02 by ivankravets