platformio/platformio-core #911: Problem with OTA upload and hostname.local

chessweb01, 2017-03-04:
I'm working with NodeMCU v1.0 and OTA firmware upload. According to the docs the upload should work with either --upload-port IP-Address or --upload-port hostname.local.

This is not the case. Using the IP-Address works without issues, using hostname.local produces the error "socket.gaierror: [Errno 11004] getaddrinfo failed"

dpeddi, 2017-03-04:
Probably isn't a platformio issue. It seems an OS issue.

ping hostname.local works?
You are using Windows? Which Release? 
If you are on Windows 7 or less have you installed  "Bonjour Print Services for Windows"?

chessweb01, 2017-03-04:
I'm on Windows 7, and I don't even know what  "Bonjour Print Services for Windows" is.

ping IP-Address works, ping hostname.local doesn't. I guess you are right, it has nothing to do with platform.io.

dpeddi, 2017-03-05:
google for "Bonjour Print Services for Windows" install and you will fix your issue.

obviously hostname.local would be what you have defined on your ArduinoOTA sketch

chessweb01, 2017-03-05:
Well, it worked. Thanks a lot. Could you elaborate a bit on the Bonjour software and the ability to resolve hostname.local?

closed 2017-03-05 by ivankravets

orcema, 2017-09-05:
For windows users instead of installing the "Bonjour Print Services for Windows" a small change has to be applied to the script **"espota.py"** located in the folder **C:\Users\[windows user name]\.platformio\packages\tool-espotapy**. 
The line of code
` remoteAddr=remoteAddr.replace(".local", "") `
has to be added according to code snippet below on line 92.
```  
# Wait for a connection
  remoteAddr=remoteAddr.replace(".local", "")
  logging.info('Sending invitation to: %s', remoteAddr)
  sock2 = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
  remote_address = (remoteAddr, int(remotePort))
  sent = sock2.sendto(message.encode(), remote_address)
  sock2.settimeout(10)```

executer-uno, 2020-06-11:
Previous comment not helps. But after revert to initial python script and installing Bonjour from
https://download.info.apple.com/Mac_OS_X/061-8098.20100603.gthyu/BonjourPSSetup.exe
It works!
Problem was on Win7 Pro and Sloeber IDE (not platformio). Default windows firewall makes no problems, can be keept running. Error message ends with following:

```
    sent = sock2.sendto(message.encode(), remote_address)
socket.gaierror: [Errno 11004] getaddrinfo failed
The execution of command "3.7.2-post1/python3" is done.
```