platformio/platformio-core #912: custom lib usage when lib_ldf_mode = off

malachib, 2017-03-06:
I am testing whether certain boost libraries (Spirit) will function ok in mbed OS environment.

These days I do all my mbed OS testing with platformio, because obviously it's awesome

I hit a snag though.  The auto LDF stuff was getting ornery so it's shut off, and now I can't figure out how to include the libs back in - specifically, the boost.config includes.  I finagled the boost.spirit into sorta including, but it seems wrong how I did it.

Here is the current not-working sample:

https://bitbucket.org/malachib/playground.mbed/src/567e23ee1033338aaa9022ecd2850421f6e299c4/src/pio/boost/?at=master

**warning** extensive use of symlinking and submodules ahead!

ivankravets, 2017-03-06:
> warning extensive use of symlinking and submodules ahead!

Please provide simple bare project that reproduces this issue.

malachib, 2017-03-07:
OK, pared down test here: https://bitbucket.org/malachib/playground.arduino/src/ab4cc11d0f4d3d10fd92592a57952ace0a5db250/PlatformIO/ldf/?at=master

ivankravets, 2017-03-08:
Take a look at https://github.com/malachi-iot/diag-collab/blob/master/src/DC-1/src/main.cpp#L4

Here is your root of a library: https://github.com/malachi-iot/diag-collab/tree/master/src/DC-1/lib/boost.config

malachib, 2017-03-09:
Indeed.  So I finagled things around with symlinks, and the **pared down test seems to compile**.  _However_, I am still facing the problem in the non-pared down project.  Things I learned:

* lib_deps seems to be sensitive to the '.' character.  I can use one, but not two in the name
* lib_deps seems sensitive to symlinking.  If in "lib/[LIBFOLDER]" the [LIBFOLDER] part itself is a symlink, it wont seem to grab the folder.  

Neither feel unreasonable to me, but it was not obvious

I need to read up more on pio libraries, because these qualify.  Except - I am proof of concepting to see if they even will compile and run in an embedded environment, so I don't want to "formalize" them, if you catch my drift?

ivankravets, 2017-03-26:
Please provide a real example how to reproduce this issue. I can't read your thoughts.

I've just tried to reproduce this issue and all works as well.

`platformio.ini`
```ini
[env:uno]
platform = atmelavr
framework = arduino
board = uno
lib_deps = ../b

```

Folder `b` is on upper level.

See PIO Core log:
```
pio run -d /tmp/111
[Sun Mar 26 21:10:49 2017] Processing uno (platform: atmelavr, lib_deps: ../b, board: uno, framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
LibraryManager: Installing b
Verbose mode can be enabled via `-v, --verbose` option
Collected 45 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <b>
|-- <a>
Compiling .pioenvs/uno/src/main.o
```

As you can see, PIO Core installed `b` library.

------

If your are going to work with LOCAL libraries, `lib_deps` is not right way here! You need use http://docs.platformio.org/en/latest/projectconf.html#lib-extra-dirs

P.S: Be sure that you use PIO Core 3.3.0a17. The final release is coming next week.

malachib, 2017-03-27:
I've been unable to reproduce it reliably.  Perhaps we should close this issue until I am able to, because it might be a while (we can always reopen the issue, right!)

I'm a little confused, because in your example I'd expect `b` library to be *local* based on the explicit pathing.  That said, I should give lib-extra-dirs a shot....

ivankravets, 2017-03-27:
Note that each entry in `lib_deps` will be passed to http://docs.platformio.org/en/latest/userguide/lib/cmd_install.html automatically.

That is why your local path in `lib_deps` PIO treats as "installation source".

closed 2017-03-27 by ivankravets