platformio/platformio-core #917: pio platform update doesn't update git dependencies

arcao, 2017-03-11:
### Configuration

**Operating system**: Windows 10 Pro x64

**PlatformIO Version** (`platformio --version`): 3.3.0a15

### Description of problem

When git dependency is updated (in my case `framework-arduinoespressif32` git dependency) platformio can't perform update for this dependency.

#### Steps to Reproduce

1. Install stage stage version with `platformio platform install https://github.com/platformio/platform-espressif32.git#feature/stage`
2. Run your project to download `framework-arduinoespressif32` git dependency from git.
3. Make change in git dependency (in my case git dependency was updated with contributors).
4. Update pio dependecies with `pio update`

### Actual Results

Git dependency is not updated.

### Expected Results

Git dependency is updated to latest commit.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:esp32dev]
platform = espressif32_stage
board = esp32dev
framework = arduino
upload_port = COM15
upload_speed = 921600
targets = upload
```

### Additional info

Solution is to reinstall `espressif32_stage` and perform run on project:

```text
>pio update
Please wait while upgrading PlatformIO...
PlatformIO has been successfully upgraded to 3.3.0a15!

*******************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for IoT development > http://platformio.org/platformio-ide
- support us with PlatformIO Plus > https://pioplus.com
*******************************************************************************

Platform Manager
================
Platform Espressif 32 (Stage)
--------
Updating espressif32_stage               @ d5458a4        [Up-to-date]
Updating tool-esptoolpy                  @ 1.20000.0      [Up-to-date]
Updating toolchain-xtensa32              @ 1.50200.0      [Up-to-date]
Updating framework-arduinoespressif32    @ 8289f34        [Up-to-date]

Library Manager
===============

>platformio platform uninstall espressif32_stage
Uninstalling espressif32_stage @ d5458a4:       [OK]
Uninstalling framework-arduinoespressif32 @ 8289f34:    [OK]
Uninstalling tool-esptoolpy @ 1.20000.0:        [OK]
Uninstalling toolchain-xtensa32 @ 1.50200.0:    [OK]
The platform 'espressif32_stage' has been successfully uninstalled!

>platformio platform install https://github.com/platformio/platform-espressif32.git#feature/stage
PlatformManager: Installing platform-espressif32
git version 2.11.1.windows.1
Cloning into 'C:\Users\Martin\.platformio\platforms\installing-gh6zxs-package'...
remote: Counting objects: 84, done.
remote: Compressing objects: 100% (59/59), done.
remote: Total 84 (delta 35), reused 48 (delta 17), pack-reused 0
Unpacking objects: 100% (84/84), done.
espressif32_stage @ d5458a4 has been successfully installed!
PackageManager: Installing toolchain-xtensa32 @ ~1.50200.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
toolchain-xtensa32 @ 1.50200.0 has been successfully installed!
PackageManager: Installing tool-esptoolpy @ ~1.20000.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
tool-esptoolpy @ 1.20000.0 has been successfully installed!
The platform 'https://github.com/platformio/platform-espressif32.git#feature/stage' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.

>pio run
[03/11/17 00:38:16] Processing esp32dev (framework: arduino, platform: espressif32_stage, upload_speed: 921600, board: esp32dev, upload_port: COM15, targets: upload)
-------------------------------------------------------------------------------
PackageManager: Installing framework-arduinoespressif32
git version 2.11.1.windows.1
Cloning into 'C:\Users\Martin\.platformio\packages\installing-bbqkvy-package'...
remote: Counting objects: 1209, done.
remote: Compressing objects: 100% (1028/1028), done.
remote: Total 1209 (delta 137), reused 930 (delta 112), pack-reused 0
Receiving objects: 100% (1209/1209), 21.02 MiB | 400.00 KiB/s, done.
Resolving deltas: 100% (137/137), done.
Checking out files: 100% (1068/1068), done.
Verbose mode can be enabled via `-v, --verbose` option
Collected 10 compatible libraries
Looking for dependencies...
Project does not have dependencies
...

>pio update
Platform Manager
================
Platform Espressif 32 (Stage)
--------
Updating espressif32_stage               @ d5458a4        [Up-to-date]
Updating tool-esptoolpy                  @ 1.20000.0      [Up-to-date]
Updating toolchain-xtensa32              @ 1.50200.0      [Up-to-date]
Updating framework-arduinoespressif32    @ 2f5efed        [Up-to-date]

Library Manager
===============
```

arcao, 2017-03-11:
Today I perform `pio update` and it downloaded latest `framework-arduinoespressif32`. So it looks like sometimes it works, sometimes not. I don't have exact step to reproduce.

ivankravets, 2017-03-11:
Thanks! Fixed in a16. Please `pio upgrade`

closed 2017-03-11 by ivankravets

emylyano3, 2018-09-20:
Is there a way to update project libs dependencies that are linked to git.
I've a lib dep in my project pointing to a git repo, like this: lib_deps=https://github.com/my-repo/my-lib.git

After pushing some changes on the lib repo the only way I see those changes in my project is deleting manually the lib folder in .piolibdeps and doing a pio run.

Does exist any other cleaner way to accomplish this?

Thanks in advance.

ivankravets, 2018-09-20:
@emylyano3 

* GUI: PlatformIO IDE for VSCode > PIO Home > Libraries > Updates
* CLI: http://docs.platformio.org/en/latest/userguide/lib/cmd_update.html

emylyano3, 2018-09-20:
@ivankravets sorry about that doc misreading. That is what I was looking for. Regards.

commit f39cda04 (2017-03-11): Fix issue when check updates for VCS repository // Resolve #917