platformio/platformio-core #921: Add framework versioning

az0uz, 2017-03-23:
This is a feature request to add framework versioning.
In the platform.ini file, it is already possible to specify the version number via semver (related #410 #413) but only for the platform and libraries, not for the framework version.

I'll explain why I need it:
I am using an atmega 32u4 with the arduino framework, and I need to use the watchdog witch is not yet supported on this board but the work is in progress as seen in:
- https://github.com/arduino/Arduino/pull/6055
- https://github.com/arduino/Arduino/issues/6033
- https://github.com/arduino/Arduino/pull/6064

to be able to still use the watchdog without waiting for this feature to be implemented, I am patching the arduino framework by:
1. using a project specific `home_dir` to not mess with the other project
2. using an `extra_script` with a `env.AddPreAction("$BUILD_DIR/FrameworkArduino/", apply_watchdog_mod)` to check/apply the watchdog patch
3. testing in my function `apply_watchdog_mod ` if the patch was applied and if not apply it.

This works great but I wouldn't be confident using that in production without being able to depend on a specific arduino framework version. If I try to install my code a new system, and the framework is updated with a new version of the files I am patching, the patch will fail.
So I would like to be able to depend on a specific framework version and upgrade everything myself when the feature is implemented in the official version of the framework.

ivankravets, 2017-03-25:
> framework is updated with a new version of the files I am patching, the patch will fail.

Framework is a part of a development platform. You can fork existing development platform and replace Arduino part with yours. See 

- http://docs.platformio.org/en/latest/platforms/creating_platform.html#manifest-file-platform-json

ivankravets, 2017-04-15:
Please reopen if you still have any questions

closed 2017-04-15 by ivankravets