platformio/platformio-core #924: platformio: command not found (from Platformio IDE)

stelgenhof, 2017-03-29:
What kind of issue is this?
- [x ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

------------------------------------------------------------------

### Configuration
Arch Linux

**PlatformIO Version** (`platformio --version`):
3.3.0

### Description of problem
When running the command "Upgrade PlatformIO Core" from the PlatformIO menu, I get the error: platformio: command not found.

This only occurred after upgrading to version 3.3.0. Before that I didn't have any issues running this option.

ivankravets, 2017-03-29:
Please open PIO IDE Terminal and run these commands:
```
whereis pio
which pio
```

stelgenhof, 2017-03-29:
I have already tried that, however said neither of these binaries were found. Actually when I ran the command "Install Shell Commands", all was fine again.

It is somewhat strange, as I never had to issue these shell commands during any update. (Perhaps only the first time when I installed PlatformIO).

ivankravets, 2017-03-29:
Could you provide `echo $PATH` from PIO IDE Terminal?

stelgenhof, 2017-03-29:
Sure. From the PIO IDE Terminal the output is:

`/usr/local/sbin:/usr/local/bin:/usr/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/home/sacha/.config/composer/vendor/bin`

ivankravets, 2017-03-30:
I see the problem. Please open new issue here https://github.com/platformio/platformio-atom-ide/issues and make reference to this issue.

The problem linked with PIO IDE and its Terminal. Before opening PIO IDE Terminal, PIO IDE passes an updated $PATH. However, it doesn't work in your case. Strange :(

closed 2017-03-30 by ivankravets

stelgenhof, 2017-03-30:
Thanks @ivankravets! I have created a new issue. As said, I resolved it by running the command "Install Shell Commands". But I guess there is some thing else at hand :)

ivankravets, 2017-03-30:
>  But I guess there is some thing else at hand :)

The core logic of PIO IDE is to use built-in PIO Core. In your case, you just installed IDE's PIO Core globally to a system.

This is good, or even, that is a right solution if you want to work with PIO CLI outside IDE. However, for me is more important to know why PIO IDE Terminal doesn't handle $PATH from PIO IDE.

What is your distro? I'll try in VirtualBox to reproduce this issue.

stelgenhof, 2017-03-30:
I am using a Linux laptop running Arch Linux. Now that you mention it, I believe I also installed PIO CLI globally after installing the IDE in Atom. I can't remember anymore why I did that, unfortunately. Actually I also prefer to use the built-in PIO Core.

ivankravets, 2017-04-30:
See https://github.com/platformio/platformio-atom-ide/issues/333#issuecomment-292252821