platformio/platformio-core #925: LibraryManager: Cloning from private git repos fails

stelgenhof, 2017-03-30:
What kind of issue is this?
- [x] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

------------------------------------------------------------------

### Configuration

**Operating system**:
Arch Linux

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 3.3.0

### Description of problem
I have a separate library stored in a private Git repository (via SSH). When adding this to the platformio.ini configuration and/or updating my project, the git clone process fails with the following error message: 

`fatal: No path specified. See 'man git-pull' for valid url syntax`

#### Steps to Reproduce
Add library dependency to the platformio.ini as follows:

`git+ssh://<username>@<hostname>:<portnumber>/somepath`

Note: I use a non-standard SSH port for accessing my git repos via SSH.

Run the build.

### Actual Results

`fatal: No path specified. See 'man git-pull' for valid url syntax`
`CalledProcessError: Command '['git', 'clone', '--recursive', '--depth', '1', 'ssh://<username>', '/path/to/project/.piolibdeps/_tmp_installing-a6aa7_-package']' returned non-zero exit status 128`

From this last part of the error output, It looks like the full path is truncated after the '@' sign. Expecting to see a format like: `ssh://<username>@<hostname>:<portnumber>/somepath`

ivankravets, 2017-03-30:
Could you temporary try this format?

- https://github.com/platformio/platformio-core/blob/develop/tests/test_managers.py#L119

stelgenhof, 2017-03-30:
Ok. Let me try that format.

stelgenhof, 2017-03-30:
Looking at that format, that is actually not how my git server is set up. My server is set up exactly as described here: https://git-scm.com/book/en/v2/Git-on-the-Server-Setting-Up-the-Server

Found out it works fine when the user part is removed from the path. So instead of this:

`git+ssh://<username>@<hostname>:<portnumber>/somepath`

I used this:

`git+ssh://<hostname>:<portnumber>/somepath`

Which makes actually sense since I am using SSH public keys so a username isn't required.

commit 1c5b08de (2017-06-24): Fixed cloning a package (library) from a private Git repository with custom user name and SSH port // Resolve #925
closed 2017-06-24 by ivankravets