platformio/platformio-core #926: Send text to Serial Port

formulahendry, 2017-03-31:
Arduino IDE and Deviot could send text to Serial Port. It seems like platformio-core cli dose not support that. I plan to add this function into the [PlatformIO VSCode extension](https://marketplace.visualstudio.com/items?itemName=formulahendry.platformio). Not sure whether this function will be added in platformio-core cli in the near future, otherwise I may want to use other library to implement that. 😄

ivankravets, 2017-04-01:
Please sorry, what do you mean? Have you seen `pio device monitor`?

formulahendry, 2017-04-01:
Oh, really sorry. I have tried `pio device monitor` before but did not find it support to send text when it is running. Thanks for pointing out!

cybergogo, 2018-11-28:
Hi, i'm looking for the same basic option that Arduino IDE offers. Have tried to find that in the pio device monitor section but nothing seems to look like what we need, which is to pass commands to our devices (arduino or ESPs):

`$ platformio device monitor

--- Available ports:
--- /dev/cu.Bluetooth-Incoming-Port n/a
--- /dev/cu.Bluetooth-Modem n/a
--- /dev/cu.SLAB_USBtoUART CP2102 USB to UART Bridge Controller
--- /dev/cu.obd2ecu-SPPDev n/a
Enter port name:/dev/cu.SLAB_USBtoUART
--- Miniterm on /dev/cu.SLAB_USBtoUART: 9600,8,N,1 ---
--- Quit: Ctrl+C  |  Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Hello PlatformIO!
---
--- Ctrl+]   Exit program
--- Ctrl+T   Menu escape key, followed by:
--- Menu keys:
---    Ctrl+T  Send the menu character itself to remote
---    Ctrl+]  Send the exit character itself to remote
---    Ctrl+I  Show info
---    Ctrl+U  Upload file (prompt will be shown)
--- Toggles:
---    Ctrl+R  RTS          Ctrl+E  local echo
---    Ctrl+D  DTR          Ctrl+B  BREAK
---    Ctrl+L  line feed    Ctrl+A  Cycle repr mode
---
--- Port settings (Ctrl+T followed by the following):
---    p          change port
---    7 8        set data bits
---    n e o s m  change parity (None, Even, Odd, Space, Mark)
---    1 2 3      set stop bits (1, 2, 1.5)
---    b          change baud rate
---    x X        disable/enable software flow control
---    r R        disable/enable hardware flow control
--- exit ---`

Maybe i'm missing something, help would be greatly appreciated.

ivankravets, 2018-11-28:
@cybergogo please note that @pyserial currently does not support sending bulk of data. See my issue/feature request https://github.com/pyserial/pyserial/issues/388

We plan to add "Serial Monitor" and other instruemnts to PIO Home soon. A temporary solution is to use free  tool Coolterm http://freeware.the-meiers.org/

cybergogo, 2018-11-29:
> @cybergogo please note that @pyserial currently does not support sending bulk of data. See my issue/feature request [pyserial/pyserial#388](https://github.com/pyserial/pyserial/issues/388)
> 
> We plan to add "Serial Monitor" and other instruemnts to PIO Home soon. A temporary solution is to use free tool Coolterm http://freeware.the-meiers.org/

Thanks for the reply. I With hope it gets integrated asap, it would help us a loooooot :-)

glrs, 2019-06-01:
@cybergogo I come a bit late to that, but I just faced this issue. Eventually, I was able to send data to my arduino through the pio serial monitor. I did so, by pressing ctrl+t (opening the menu), followed by another ctrl+t. Then I just typed what I wanted to send and hit enter. It worked. However, I could not see the command I was typing. This is still a bad design and error prone.

@ivankravets how is things with this issue?

ivankravets, 2019-06-01:
Depends on https://github.com/pyserial/pyserial/issues/388

ivankravets, 2020-03-17:
Please upgrade to the latest version via `pio upgrade --dev` and re-test with `send_on_enter` filter. See docs https://docs.platformio.org/en/latest/core/userguide/device/cmd_monitor.html

Does it work?

closed 2020-03-17 by ivankravets

brianfaires, 2020-04-04:
Working for me.  Would love to see the text as it's typed, but send_on_enter working in VScode pio plugin.

ivankravets, 2020-04-04:
Does it work if you use `--echo` flag in a combination with filters?

brianfaires, 2020-04-04:
Yes, echo flag works!  This is my first week with PIO, I can't believe how long I was using the Arduino IDE...  Thanks for the quick responses and contributions!

commit 605cd36e (2020-03-17): Send a text to device on ENTER with ``send_on_enter`` filter // Resolve #926