platformio/platformio-core #928: xtensa-lx106-elf-gcc is not a known internal or external command ...

FernandoGarcia, 2017-04-01:
Hi!

I have installed the IDE yesterday and I'm trying compile "project examples" for ESP8266 but I have always this error:

```
Compiling .pioenvs\esp01\src\blinky.o
'xtensa-lx106-elf-gcc' não é reconhecido como um comando interno
ou externo, um programa operável ou um arquivo em lotes.
*** [.pioenvs\esp01\src\blinky.o] Error 1
 [ERROR] Took 1.63 seconds
```

This result is for ESP8266-native sketch.

All packets are up to date.

```
PS C:\Users\Fernando Garcia\.platformio\project-examples\espressif\esp8266-native> platformio update
Updating tool-scons                      @ 3.20501.2      [Up-to-date]
 
Platform Manager
================
Platform Espressif 32
--------
Updating espressif32                     @ 0.6.0          [Up-to-date]
Updating tool-esptoolpy                  @ 1.20000.0      [Up-to-date]

Updating toolchain-xtensa32              @ 1.50200.0      [Up-to-date]
Updating framework-arduinoespressif32    @ 1.1.2          [Up-to-date]
 
Platform Espressif 8266
--------
Updating espressif8266                   @ 1.3.0          [Up-to-date]
Updating framework-arduinoespressif8266  @ 1.20300.1      [Up-to-date]
Updating sdk-esp8266                     @ 1.10502.0      [Up-to-date]
Updating tool-esptool                    @ 1.409.0        [Up-to-date]
Updating toolchain-xtensa                @ 1.40802.0      [Up-to-date]
 
 
Library Manager
===============
PS C:\Users\Fernando Garcia\.platformio\project-examples\espressif\esp8266-native> platformio --version
PlatformIO, version 3.3.0
```

This error is for ESP8266-webserver sketch:

[image]

Can you help me please?

Moved from: https://github.com/platformio/platformio-atom-ide/issues/307

Best regards.

douglaslaraa, 2017-04-02:
Friend, try to uninstall the platformio, remove the ".platformio" folder from the home directory if you use linux and install the platformio again.

FernandoGarcia, 2017-04-24:
Obrigado @douglaslaraa !

I'm using Win 10 and the path to `.platformio` is: C:\Users\Username\.platformio

closed 2017-04-24 by FernandoGarcia