platformio/platformio-core #933: Preprocessor bug ignores #ifndef - breaks Blink example!

n0dyjeff, 2017-04-09:
What kind of issue is this?

- [X] PlatformIO Core. If you’ve found a bug, please provide an information below.

### Configuration

Windows 10

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 3.4.0a4

### Description of problem
**Summary**: An #ifndef preprocessor conditional is ignored, causing the conditionally included statements to be compiled regardless of the fact that the symbol exists.

**Details**: I was testing my new installation of PlatformIO IDE, and followed the steps to build the Blink example. I was astonished to find that it didn't work. What I found was the following code ignored the definition of LED_BUILTIN (found in pins_arduino.h, included by Arduino.h and set to 16) and instead redefined it to 13.
```
#ifndef LED_BUILTIN
#define LED_BUILTIN 13
#endif
```
This code caused GPIO 13 to be toggled - which is incorrect, since the LED_BUILTIN value is 16. Changing the code to comment out the #define corrects the issue:
```
#ifndef LED_BUILTIN
//#define LED_BUILTIN 13
#endif
```
#### Steps to Reproduce

1. Build and install the source code below to a NodeMCU V1.0. Note serial output and behavior of built in LED.
2. Comment out the LED_BUILTIN definition and again note serial output and built-in LED behavior.
```
#ifndef LED_BUILTIN
//#define LED_BUILTIN 13
#endif
```
### Actual Results
**Step 1**
Serial output:
`LED_BUILTIN = 13`
Built in LED does not blink (but GPIO 13 toggles)
**Step 2**
Serial output:
`LED_BUILTIN = 16`
Built in LED blinks

Note that if LED_BUILTIN was _truly_ undefined (as indicated by the results of Step 1), there should have been a compile failure due to the symbol not being found. 
### Expected Results
**Step 1**
Serial output:
`LED_BUILTIN = 16`
Built in LED blinks
**Step 2**
Serial output:
`LED_BUILTIN = 16`
Built in LED blinks

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
upload_speed = 460800
```

**Source file to reproduce issue:**
```cpp
#include "Arduino.h"

#ifndef LED_BUILTIN
#define LED_BUILTIN 13
#endif

void setup()
{
  // initialize LED digital pin as an output.
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(115200);
}

void loop()
{
  // turn the LED on (HIGH is the voltage level)
  digitalWrite(LED_BUILTIN, HIGH);
  Serial.print("LED_BUILTIN = ");
  Serial.println(LED_BUILTIN);
  // wait for a second
  delay(1000);

  // turn the LED off by making the voltage LOW
  digitalWrite(LED_BUILTIN, LOW);

   // wait for a second
  delay(1000);
}
```

### Additional info

ivankravets, 2017-04-15:
PIO Core doesn't touch your code. Do you use `*.cpp` extension? I've just even tried to debug and prepared for you real-time picture. As you can see, I overrode `LED_BUILTIN` with `15` but `13` is the actual value. See watch pointer for `ulPin`:

[image]

n0dyjeff, 2017-04-17:
Hi Ivan, thanks for looking in to this.

Yes, my file is named main.cpp

I'm not sure I understand your results. You are getting a value of 13 for `LED_BUILTIN`. That isn't the value defined in the NodeMCU version of "`pins_arduino.h`", which is 16. Were you building against "`nodemcuv2`"? If so, I don't understand where the value of 13 is coming from for you. It should be 16.

Regardless, I see your point that `LED_BUILTIN` is not being redefined to 15 when you run your code. So in your case, it's behaving as though the symbol is defined and skipping over the `#define`.

So, we are running similar tests but getting different results. There must be some differences that account for this.

If you would share your source code, I will run it on my system and see what results I get. Perhaps that will help us get to the root of the issue.

n0dyjeff, 2017-04-18:
Now that I look more carefully, I see that you are just setting watchpoints to capture the value passed to pinMode.  So I guess I don't need your source code :).  But I'm a noob to Platform.io, so it may take me a while to figure out how to do watchpoints for a NodeMCU build - if that's even possible. I suspect that I might have to build for the host platform (Windows) to use? I'll find out.

Is it possible you could try to reproduce the problem by downloading to a NodeMCU, as I show in my original note? It would be good to know if someone else gets this same behavior, or if it's unique somehow to my installation.

ivankravets, 2017-04-27:
Do you use `.cpp` for `.ino` format?

n0dyjeff, 2017-04-27:
My file is named main.cpp, exactly as shown in the [example application](http://docs.platformio.org/en/latest/ide/atom.html#quick-start).

ivankravets, 2017-04-27:
Thanks, I reproduced this issue!

n0dyjeff, 2017-04-27:
Awesome! Thanks for staying with it.

ivankravets, 2017-04-27:
The problem that in Arduino for Espressif `LED_BUILTIN` is set as static variable(not macro). See  https://github.com/esp8266/Arduino/blob/master/variants/nodemcu/pins_arduino.h#L37

I'll update tutorial. Please remove that `#ifdef ` block.

n0dyjeff, 2017-04-27:
OK, thanks very much. Again, many thanks for digging into it. I was having difficulty committing to move forward with Platformio when such a simple example didn't work. Now I can adopt it with greater confidence.

ivankravets, 2017-04-27:
All examples have been updated! Thank you too!

commit 81c96808 (2017-04-27): Don’t override LED_BUILTIN with macro; sometime it is defined as static variable // Resolve #933
closed 2017-04-27 by ivankravets