platformio/platformio-core #937: Can't find toolchains for aarch64 host system

hoosierEE, 2017-04-11:
What kind of issue is this?

- [x] PlatformIO Core. If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Ubuntu 16.04.2 LTS

**PlatformIO Version** (`platformio --version`): 3.3.0

### Description of problem

Note: my computer has an aarch64 (ARM v8) processor architecture.

Attempting to run platformio results in "toolchain not found" errors.

#### Steps to Reproduce

1. install Ubuntu
2. install pip
3. use "super quick" install (`python -c "$(curl ... `)
4. write `platformio.ini` file and some source code under `src/main.cpp` (given below)
5. `pio run`

### Actual Results
`Error: Could not find a version that satisfies the requirement '~1.40902.0' for your system 'linux_aarch64'`

### Expected Results
Build finishes without error.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:arduino]
platform = atmelavr
board = nanoatmega328
framework = arduino
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>
void setup(){}
void loop(){}
```

### Additional info

My laptop has an ARMv8 (aarch64) processor.

This bug is repeatable with the teensy toolchain as well

```ini
[env:teensylc]
platform = teensy
board = teensylc
framework = arduino
```

Error message:

`Error: Could not find a version that satisfies the requirement '>=1.40803.0,<1.40803.0' for your system 'linux_aarch64'`

ivankravets, 2017-04-15:
All our ARM-based packages are compiled for ARMv6/v7. I don't know if they are compatible with ARMv8.  Could you try manually download some of them and check if ARM keeps compatibility between v7 & v8. If they work, I'll re-pack all ARM-related package and allow `aarch64`.

1. Please download this archive and unpack it https://bintray.com/platformio/dl-packages/toolchain-gccarmnoneeabi/1.40803.0#files
2. Run `bin/arm-none-eabi-g++ -v` from unpacked directory.

Does it work?

hoosierEE, 2017-04-15:
ARMv8 should be able to run ARMv7 code without any modification.  Anyway, running `arm-none-eabi/bin/g++ -v` gives me this result:

```bash
lx@~/Downloads/arm-none-eabi/bin g++ -v
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/5/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.4' --with-bugurl=file:///usr/share/doc/gcc-5/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-5 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-5-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-5-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-5-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --enable-multilib --disable-sjlj-exceptions --with-arch=armv7-a --with-fpu=vfpv3-d16 --with-float=hard --with-mode=thumb --disable-werror --enable-multilib --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 5.4.0 20160609 (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.4) 
```

hoosierEE, 2017-04-15:
Oops, my apologies.  I did not notice the top-level `bin/` directory at first.  Here is the output of the command you requested, exactly:

```
lx@~/Downloads bin/arm-none-eabi-g++ -v
Using built-in specs.
COLLECT_GCC=bin/arm-none-eabi-g++
COLLECT_LTO_WRAPPER=/home/lx/Downloads/bin/../lib/gcc/arm-none-eabi/4.8.3/lto-wrapper
Target: arm-none-eabi
Configured with: /home/paul/ARM_Toolchain_2014q1_Source/src/gcc/configure --target=arm-none-eabi --prefix=/home/paul/ARM_Toolchain_2014q1_Source/install-native --libexecdir=/home/paul/ARM_Toolchain_2014q1_Source/install-native/lib --infodir=/home/paul/ARM_Toolchain_2014q1_Source/install-native/share/doc/gcc-arm-none-eabi/info --mandir=/home/paul/ARM_Toolchain_2014q1_Source/install-native/share/doc/gcc-arm-none-eabi/man --htmldir=/home/paul/ARM_Toolchain_2014q1_Source/install-native/share/doc/gcc-arm-none-eabi/html --pdfdir=/home/paul/ARM_Toolchain_2014q1_Source/install-native/share/doc/gcc-arm-none-eabi/pdf --enable-languages=c,c++ --enable-plugins --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-newlib --with-headers=yes --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/home/paul/ARM_Toolchain_2014q1_Source/install-native/arm-none-eabi --with-gmp=/home/paul/ARM_Toolchain_2014q1_Source/build-native/host-libs/usr --with-mpfr=/home/paul/ARM_Toolchain_2014q1_Source/build-native/host-libs/usr --with-mpc=/home/paul/ARM_Toolchain_2014q1_Source/build-native/host-libs/usr --with-isl=/home/paul/ARM_Toolchain_2014q1_Source/build-native/host-libs/usr --with-cloog=/home/paul/ARM_Toolchain_2014q1_Source/build-native/host-libs/usr --with-libelf=/home/paul/ARM_Toolchain_2014q1_Source/build-native/host-libs/usr --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --with-pkgversion='GNU Tools for ARM Embedded Processors' --with-multilib-list=armv6-m,armv7-m,armv7e-m,armv7-r
Thread model: single
gcc version 4.8.3 20140228 (release) [ARM/embedded-4_8-branch revision 208322] (GNU Tools for ARM Embedded Processors) 
```

ivankravets, 2017-04-15:
Could you try again to build Teensy project?

hoosierEE, 2017-04-15:
Same error.  Also I put the `bin/` folder on my PATH and again same error.

ivankravets, 2017-04-15:
No need to put in PATH PIO packages. PIO does that automatically in a virtual build environment. Please remove PIO packages from PATH ... and remove downloaded packages.

I see the error, I updated packages with 'aarch64` instead of `linux_aarch64` :(

I'll re-deploy all packages and inform here.

hoosierEE, 2017-04-15:
Success, thanks Ivan!

ivankravets, 2017-04-16:
Does even work uploading?

ivankravets, 2017-04-16:
What is your machine with this `linux_aarch64`?

hoosierEE, 2017-04-17:
Haven't tried uploading yet, will update this thread with the results soon.

I'm using an Acer Chromebook R13 ( affiliate [link](http://amzn.to/2prwNzZ) ), with [Crouton](https://github.com/dnschneid/crouton/) to manage the Ubuntu/Debian environment.

hoosierEE, 2017-04-17:
Update.  There is an issue with the upload process, here's the output from `pio run -v`:

```
lx@~/Downloads/blinky-light-strip/pio[master !?*] pio run -v
[Mon Apr 17 09:52:39 2017] Processing teensylc (targets: upload, platform: teensy, board: teensylc, framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------
Collected 91 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <SPI> v1.0 (/home/lx/.platformio/packages/framework-arduinoteensy/libraries/SPI)
|-- <LPD8806> v1.0.0 (/home/lx/Downloads/blinky-light-strip/pio/lib/LPD8806)
|   |-- <SPI> v1.0 (/home/lx/.platformio/packages/framework-arduinoteensy/libraries/SPI)
arm-none-eabi-g++ -o .pioenvs/teensylc/firmware.elf -Os -Wl,--gc-sections,--relax -mthumb -mcpu=cortex-m0plus -Wl,--defsym=__rtc_localtime=1492437159 -fsingle-precision-constant --specs=nano.specs -Wl,-T"mkl26z64.ld" .pioenvs/teensylc/src/main.o -L/home/lx/.platformio/packages/framework-arduinoteensy/cores/teensy3 -L.pioenvs/teensylc -Wl,--start-group .pioenvs/teensylc/libFrameworkArduino.a -lm -lc -lgcc -larm_cortexM0l_math .pioenvs/teensylc/lib/libSPI.a .pioenvs/teensylc/lib/libLPD8806.a -Wl,--end-group
Checking program size
text       data     bss     dec     hex filename
13768       292    2068   16128    3f00 .pioenvs/teensylc/firmware.elf
arm-none-eabi-objcopy -O ihex -R .eeprom .pioenvs/teensylc/firmware.elf .pioenvs/teensylc/firmware.hex
teensy_loader_cli -mmcu=mkl26z64 -w -s -v .pioenvs/teensylc/firmware.hex
Teensy Loader, Command Line, Version 2.1
Read ".pioenvs/teensylc/firmware.hex": 14060 bytes, 22.1% usage
Found HalfKay Bootloader
error writing to Teensy

*** [upload] Error 1
Programming.
======================================================== [ERROR] Took 5.80 seconds ========================================================
```

hoosierEE, 2017-04-17:
Almost there!

Using the latest `teensy_loader_cli` (downloaded source from GitHub and built on the aarch64 machine) I can upload the (PlatformIO-generated) hex file to my device, and it works:

```
lx@~/Downloads/blink ../teensy_loader_cli/teensy_loader_cli -v -w -mmcu=mk64fx512 .pioenvs/teensy35/firmware.hex 
Teensy Loader, Command Line, Version 2.1
Read ".pioenvs/teensy35/firmware.hex": 17032 bytes, 3.2% usage
Waiting for Teensy device...
 (hint: press the reset button)
Found HalfKay Bootloader
Read ".pioenvs/teensy35/firmware.hex": 17032 bytes, 3.2% usage
Programming.................
Booting
```

This is using `teensy_loader_cli` at revision [f5b6d7a](https://github.com/PaulStoffregen/teensy_loader_cli/tree/f5b6d7aafda9a8b014b4bb08660833ca45c136d2).

Hopefully this is easy to incorporate this latest version into PlatformIO.

ivankravets, 2017-04-27:
Plese open new issue here https://github.com/platformio/platform-teensy/issues and report that need to recompile `teensy_loader_cli` for ARM.

Thanks!

closed 2017-04-27 by ivankravets