platformio/platformio-core #943: Re-Return of #48 - Problem with #include <SPI.h>

Nzbuu, 2017-04-19:
Using PlatformIO v3.4.0a6 (which is the current PIP version), I get the following error, unless I explicitly include SPI in the dependencies or in the main CPP-file.

```
fatal error: SPI.h: No such file or directory
```

My understanding of a comment in #702 [1] is that it should not be necessary to explicitly include this  in v3. Is this the intention, or is it still necessary to include the dependency explicitly?

[1] https://github.com/platformio/platformio-core/issues/702#issuecomment-227011519

**The content of `platformio.ini`:**
```ini
[env:uno]
platform = atmelavr
board = uno
framework = arduino
lib_deps =
  U8g2
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>
#include <U8g2lib.h>

void setup(void) {
}

void loop(void) {
}
```

ivankravets, 2017-04-20:
See http://docs.platformio.org/en/latest/librarymanager/ldf.html#dependency-finder-mode

So, if `U8g2lib.h` doesn't contain `#include <SPI.h>` directly, then you have to add it to your project or change dependency finder mode. I recommend to add `#include <U8g2lib.h>` to project:

```cpp
#include <Arduino.h>
#include <SPI.h>
#include <U8g2lib.h>

void setup(void) {
}

void loop(void) {
}
```

Nzbuu, 2017-04-24:
OK. Thanks.

ivankravets, 2017-04-24:
Thanks, added!

https://github.com/platformio/platformio-docs/compare/310c542680b833866ef3256bd0b28ea763e1a9cd...df9accd13b4c4a9cf379b09e8b032c61b36aab43#diff-a5f3903f22caf22a7edc29548574191b

closed 2017-04-24 by ivankravets