platformio/platformio-core #944: only parse platformio.ini when necessary

boneskull, 2017-04-20:
- [x] PlatformIO Core

**PlatformIO Version** (`platformio --version`):

`3.3.0`

### Description of problem

When in a project directory, `pio` exits with an exception if `platformio.ini` is unparseable, *regardless* of the command run.

#### Steps to Reproduce

1.  Create a project with `pio init`
2.  Add a library dependency in `platformio.ini` as you would normally
3.  Remove the indent from the library dependency, e.g.

    ```ini
    [env:esp01]
    platform = espressif8266
    board = esp01
    framework = arduino
    lib_deps =
    AsyncMqttClient@^0.5.0
    ```
4. Execute `pio lib search <something>`

### Actual Results

```
Traceback (most recent call last):
  File "/usr/local/bin/pio", line 11, in <module>
    load_entry_point('platformio==3.3.0', 'console_scripts', 'pio')()
  File "/usr/local/Cellar/platformio/3.3.0/libexec/lib/python2.7/site-packages/pkg_resources/__init__.py", line 560, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/local/Cellar/platformio/3.3.0/libexec/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2648, in load_entry_point
    return ep.load()
  File "/usr/local/Cellar/platformio/3.3.0/libexec/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2302, in load
    return self.resolve()
  File "/usr/local/Cellar/platformio/3.3.0/libexec/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2308, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/Cellar/platformio/3.3.0/libexec/lib/python2.7/site-packages/platformio/__main__.py", line 24, in <module>
    from platformio import __version__, exception, maintenance
  File "/usr/local/Cellar/platformio/3.3.0/libexec/lib/python2.7/site-packages/platformio/maintenance.py", line 25, in <module>
    from platformio.commands.lib import lib_update as cmd_lib_update
  File "/usr/local/Cellar/platformio/3.3.0/libexec/lib/python2.7/site-packages/platformio/commands/lib.py", line 37, in <module>
    " library storage `%s`" % join(util.get_home_dir(), "lib"))
  File "/usr/local/Cellar/platformio/3.3.0/libexec/lib/python2.7/site-packages/platformio/util.py", line 214, in get_home_dir
    join(expanduser("~"), ".platformio"))
  File "/usr/local/Cellar/platformio/3.3.0/libexec/lib/python2.7/site-packages/platformio/util.py", line 194, in get_project_optional_dir
    config = load_project_config()
  File "/usr/local/Cellar/platformio/3.3.0/libexec/lib/python2.7/site-packages/platformio/util.py", line 314, in load_project_config
    cp.read(path)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ConfigParser.py", line 305, in read
    self._read(fp, filename)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ConfigParser.py", line 546, in _read
    raise e
ConfigParser.ParsingError: File contains parsing errors: /Volumes/alien/platformio/bee-em-pee/platformio.ini
```

### Expected Results

In this case, would expect to see the results of my search.

### Proposed Solution

Searching for libraries, checking the version (`pio --version`), and likely a host of other commands should not need to check for valid syntax in `platformio.ini`.  

The list of commands that require parsing of the `ini` file should be enumerated, and the code should be refactored so the commands *not* on this list do not parse the `ini` file.

### Workaround

`cd` to a different directory w/o an invalid `platformio.ini` file, or fix your `platformio.ini` file.  :smile:

ivankravets, 2017-04-20:
```
lib_deps =
  AsyncMqttClient@^0.5.0
```

Multi-line values should start minimum from 2 spaces.

boneskull, 2017-04-20:
@ivankravets Yes, that is invalid syntax, but it's beside the point of this ticket.  I don't think this is invalid.  Can you please re-open?  Not sure if you read it closely.

boneskull, 2017-04-20:
(In other words: `pio` should not die w/ an exception for commands that have nothing to do with the contents of `platformio.ini`)

boneskull, 2017-04-20:
thanks :smile:

ivankravets, 2017-04-21:
`platformio.ini` is needed here to check if `lib_dir` is overwritten. It will change behaviour of Library Manager.

closed 2017-04-27 by ivankravets