platformio/platformio-core #945: ESP32 OTA can't upload to IP.

GanjaFarmer, 2017-04-21:
Hello,

I received my ESP32 dev board, I'm using Arduino ESP32 package and try to do OTA upload.

First it couldn't find some libraries, so I tried to manually install them by downloading the .zip from https://github.com/espressif/arduino-esp32 and placing content in ".platformio\packages\framework-arduinoespressif32\libraries". Still an error, so after a few searches I found that I needed to use
'''platform = espressif32_stage'''
in the platfformio.ini.

Now the errors are fixed, so I added
'''upload_port = 192.168.0.17'''

But it gives me these errors whenever I try to Upload:
'''esptool.py v2.0-beta1

Looking for upload port...
Use manually specified: 192.168.0.17
Uploading .pioenvs\esp32dev\firmware.bin
esptool.py v2.0-beta2
Traceback (most recent call last):
File "C:\Users\Guillaume\.platformio\packages\framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6\tools\esptool.py", line 2294, in <module>
main()
File "C:\Users\Guillaume\.platformio\packages\framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6\tools\esptool.py", line 2057, in main
esp = chip_class(args.port, initial_baud)
File "C:\Users\Guillaume\.platformio\packages\framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6\tools\esptool.py", line 166, in __init__
self._port = serial.serial_for_url(port)
File "c:\users\guillaume\.platformio\penv\lib\site-packages\serial\__init__.py", line 88, in serial_for_url
instance.open()
File "c:\users\guillaume\.platformio\penv\lib\site-packages\serial\serialwin32.py", line 62, in open
raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port '192.168.0.17': WindowsError(2, 'Le fichier sp\xe9cifi\xe9 est introuvable.')
*** [upload] Error 1
 [ERROR] Took 12.22 seconds'''

The IP is correct, I installed Bonjour... I don't know what to try next. Am the only one having this problem ?

PS: I've just installed Windows 10, could it be the issue ? I didn't have this problem with **ESP8266** OTA on Windows 8.1.

ivankravets, 2017-04-25:
Please move this issue to https://github.com/platformio/platform-espressif32/issues

closed 2017-04-25 by ivankravets

pikolo84, 2017-04-27:
I have the same problem.

ivankravets, 2017-04-27:
Moved to https://github.com/platformio/platform-espressif32/issues/22

GanjaFarmer, 2017-04-27:
Thanks ivankravets ;)