platformio/platformio-core #946: IDE can't deal with multiples ino files yet

FernandoGarcia, 2017-04-24:
I have a code working with Arduino IDE but it doesn't work with platformIO.

The issue with multiples ino files was pointed as solved [here](https://github.com/platformio/platformio-core/issues/130) but it doesn't work to me.

See the print screen with errors:

[image]

Here the full code: https://github.com/FernandoGarcia/Ferduino_with_webcontrol_beta/tree/master/Ferduino_with_webcontrol_beta

Another thing, the IDE is adding `#include <Arduino.h>` to each file on code. It's really needed?

Best regards.

ivankravets, 2017-04-24:
Please check that you don't have "space" in a new line before prototype.
For example, 
```
void correct();
 void incorrect();
```

I remember that someone reported this bug before.

ivankravets, 2017-04-24:
Also, we highly recommend switching from INO files to CPP. We are working on a debugger feature for PIO IDE 2.0 and it will not work with multiple INOs.

More info http://docs.platformio.org/en/latest/ide/atom.html#convert-arduino-file-to-c-manually

ivankravets, 2017-04-27:
@valeros could you check that?

valeros, 2017-04-27:
Hi @FernandoGarcia !
There is no an easy way to convert such a complicated project to PlatformIO. So if you are going to use PlatformIO I suggest you reorganize your project according to C/C++ guidelines (with source/header files).

FernandoGarcia, 2017-05-02:
Ok, thanks!

closed 2017-05-02 by FernandoGarcia