platformio/platformio-core #950: Exceptions

DJManas, 2017-04-27:
What kind of issue is this?

- [X] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Development Platform or Board. All issues related to Development Platforms or Embedded Boards
      should be reported to appropriate repository. 
      See full list with repositories and search for "platform-xxx" repository related to your hardware
      https://github.com/platformio?query=platform-

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**MacOS**:

**PlatformIO Version** (`platformio --version`): PlatformIO, version 3.4.0a6

I just want to ask if this is pio problem or I should contact someone else. When I create program and use e.g. A1 as analog pin number I get errors in project that A1 was not declared in this scope, but when I compile it everything is OK. I would like to know if this is pio problem or I should contact anyone else to repair it, because it is anoying.
[image]

#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

ivankravets, 2017-04-27:
Have you tried Menu: PlatformIO > Rebuild C/C++ Index ?

DJManas, 2017-04-28:
Nope :-( since I see it is happening automatically at some point. But I have tried it now, did make clean of the project, now I see only A1 is not declared, the SHT1x.h has been found and it skipped out of the scope. 

I am just wondering couldn’t it be, because the analog pins are „declared“ when at compile using #if defined(processor type) ?

ivankravets, 2017-04-28:
Could you provide `platformio.ini`?

DJManas, 2017-04-28:
Sure
[platformio.ini.zip](https://github.com/platformio/platformio-core/files/964270/platformio.ini.zip)

ivankravets, 2017-04-28:
This is a bug, PIO does not handle `env_default` from config. I'll fix it. Thanks.

commit 1e5df747 (2017-04-28): Handle ``env_default`` in "platformio.ini" when re-initializing a project // Resolve #950
closed 2017-04-28 by ivankravets