platformio/platformio-core #952: UnicodeDecodeError when generating CLion integration files

Oromis, 2017-05-02:
When setting up our PlatformIO project on my coworker's laptop, I ran into an encoding error while running `pio init --ide clion`. She has a non-ASCII character in her Windows username, which is nothing I can do anything about.

I read issue https://github.com/platformio/platformio-core/issues/143#issuecomment-88060906 and applied all suggestions.

Steps I took in an attempt to resolve the error:

* Place the project in a directory outside of her home directory (C:\Software)
* Defined home_dir in platformio.ini to point at a directory without non-ASCII characters (C:\\.platformio)
* Defined the System Environment Variable PLATFORMIO_HOME_DIR to point at the same directory (C:\\.platformio)

None of these worked. I can run pio init successfully, but the CLion IDE integration makes it fail. Here is the traceback:

```
Error: Traceback (most recent call last):
  File "c:\python27\lib\site-packages\platformio\__main__.py", line 106, in main
    cli(None, None, None)
  File "c:\python27\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\python27\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\python27\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\python27\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\python27\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\python27\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\python27\lib\site-packages\platformio\commands\init.py", line 114, in cli
    pg.generate()
  File "c:\python27\lib\site-packages\platformio\ide\projectgenerator.py", line 116, in generate
    self._render_tpl(tpl_path).encode("utf8"))
  File "c:\python27\lib\site-packages\platformio\ide\projectgenerator.py", line 122, in _render_tpl
    return bottle.template(content, **self._tplvars)
  File "c:\python27\lib\site-packages\bottle.py", line 3619, in template
    return TEMPLATES[tplid].render(kwargs)
  File "c:\python27\lib\site-packages\bottle.py", line 3409, in render
    self.execute(stdout, env)
  File "c:\python27\lib\site-packages\bottle.py", line 3396, in execute
    eval(self.co, env)
  File "<string>", line 1, in <module>
  File "c:\python27\lib\site-packages\bottle.py", line 3347, in <lambda>
    self._escape = lambda x: escape_func(touni(x, enc))
  File "c:\python27\lib\site-packages\bottle.py", line 123, in touni
    return s.decode(enc, err) if isinstance(s, bytes) else unicode(s)
  File "c:\python27\lib\encodings\utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xe4 in position 217: invalid continuation byte
```

The traceback refers to the invalid character 0xE4, which is the Windows-1252 representation of the umlaut ä, which is a part of my coworker's username.

ivankravets, 2017-05-02:
What do you see in a system terminal?
```
echo %PATH%
```

Oromis, 2017-05-03:
The Output ist: 

C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Programms\Git\cmd;C:\Python27;C:\Python27\Scripts;C:\Programms\CMake\bin;C:\MinGW\bin;C:\Users\Rebecca Kächele\AppData\Local\Microsoft\WindowsApps;

ivankravets, 2017-05-04:
Please re-test with http://docs.platformio.org/en/latest/installation.html#development-version

Oromis, 2017-05-05:
I re-tested the development version, but with the same result unfortunately.

```
Error: Traceback (most recent call last):
  File "c:\python27\lib\site-packages\platformio\__main__.py", line 106, in main
    cli(None, None, None)
  File "c:\python27\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\python27\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\python27\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\python27\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\python27\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\python27\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\python27\lib\site-packages\platformio\commands\init.py", line 103, in cli
    pg.generate()
  File "c:\python27\lib\site-packages\platformio\ide\projectgenerator.py", line 121, in generate
    self._render_tpl(tpl_path).encode("utf8"))
  File "c:\python27\lib\site-packages\platformio\ide\projectgenerator.py", line 127, in _render_tpl
    return bottle.template(content, **self._tplvars)
  File "C:\Python27\Scripts\bottle.py", line 3619, in template
    return TEMPLATES[tplid].render(kwargs)
  File "C:\Python27\Scripts\bottle.py", line 3409, in render
    self.execute(stdout, env)
  File "C:\Python27\Scripts\bottle.py", line 3396, in execute
    eval(self.co, env)
  File "<string>", line 1, in <module>
  File "C:\Python27\Scripts\bottle.py", line 3347, in <lambda>
    self._escape = lambda x: escape_func(touni(x, enc))
  File "C:\Python27\Scripts\bottle.py", line 123, in touni
    return s.decode(enc, err) if isinstance(s, bytes) else unicode(s)
  File "c:\python27\lib\encodings\utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xe4 in position 231: invalid continuation byte
```

ivankravets, 2017-05-05:
Where is located your project? Does it's path contain non-ascii chars?

Oromis, 2017-05-05:
No, the project directory is `C:\Software\<project-name>` and the project name does not contain non-ASCII chars or spaces.

ivankravets, 2017-05-06:
Please try this debug version of PIO Core:
```
pip install -U https://github.com/platformio/platformio-core/archive/feature/issue-952-UnicodeDecodeError.zip
```

Try to init project again and provide an output here. Thanks!

Oromis, 2017-05-09:
I finally got around to trying it out. Here is the output:

```
>pio init --ide clion

The current working directory C:\Software\vfm will be used for project.
You can specify another project directory via
`platformio init -d %PATH_TO_THE_PROJECT_DIR%` command.

The next files/directories have been created in C:\Software\vfm
platformio.ini - Project Configuration File
src - Put your source files here
lib - Put here project specific (private) libraries
_render_tpl {'project_name': 'vfm', u'prog_path': u'C:\\Software\\vfm\\.pioenvs\\nodemcu\\firmware.elf', 'src_files': ['app\\unity_config.h', 'app\\arduino\\wire.cpp', 'app\\arduino\\wire.h', 'app\\src\\Main.cpp', 'app\\src\\Sandbox.cpp', 'app\\src\\programs\\AdPortCheck.cpp', 'app\\src\\programs\\AdPortCheck.h', 'app\\src\\programs\\ModbusSlaveTest.cpp', 'app\\src\\programs\\ModbusSlaveTest.h', 'app\\src\\programs\\Program.h', 'app\\src\\programs\\VfmInterface.cpp', 'app\\src\\programs\\VfmInterface.h', 'app\\test\\Test.cpp', 'app\\test\\helper\\FakeStream.cpp', 'app\\test\\helper\\FakeStream.h', 'app\\test\\helper\\unity_config.cpp', 'app\\test\\unit\\AdPortCheckTest.cpp', 'app\\test\\unit\\VfmInterfaceTest.cpp', 'app\\test\\unity\\unity.c', 'app\\test\\unity\\unity.h', 'app\\test\\unity\\unity_fixture.cpp', 'app\\test\\unity\\unity_fixture.h', 'app\\test\\unity\\unity_fixture_internals.h', 'app\\test\\unity\\unity_fixture_malloc_overrides.h', 'app\\test\\unity\\unity_internals.h'], u'cxx_flags': u'-fno-rtti -fno-exceptions -std=c++11 -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -ffunction-sections -fdata-sections -DARDUINO=20300 -DLWIP_OPEN_SRC -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=30400 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -U__STRICT_ANSI__', u'includes': [u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\tools\\sdk\\include', u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\tools\\sdk\\lwip\\include', u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\cores\\esp8266', u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\variants\\nodemcu', u'C:\\Software\\vfm\\.piolibdeps\\volt-common', u'C:\\Software\\vfm\\app', u'C:\\Software\\vfm\\.piolibdeps\\volt-common', u'C:\\Software\\vfm\\.piolibdeps\\volt-common', u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\libraries\\ArduinoOTA', u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\libraries\\DNSServer\\src', u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\libraries\\EEPROM', u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\libraries\\ESP8266AVRISP\\src', u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\libraries\\ESP8266HTTPClient\\src', u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\libraries\\ESP8266HTTPUpdateServer\\src', u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\libraries\\ESP8266SSDP', u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\libraries\\ESP8266WebServer\\src', u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\libraries\\ESP8266WiFi\\src', u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\libraries\\ESP8266WiFiMesh\\src', u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\libraries\\ESP8266httpUpdate\\src', u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\libraries\\ESP8266mDNS', u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\libraries\\Ethernet\\src', u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\libraries\\GDBStub\\src', u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\libraries\\Hash\\src', u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\libraries\\SD\\src', u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\libraries\\SPI', u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\libraries\\Servo\\src', u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\libraries\\SoftwareSerial', u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\libraries\\TFT_Touch_Shield_V2', u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\libraries\\Ticker', u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\libraries\\Wire', u'C:\\.platformio\\packages\\framework-arduinoespressif8266\\libraries\\esp8266\\src', u'C:\\.platformio\\packages\\toolchain-xtensa\\xtensa-lx106-elf\\include', u'C:\\.platformio\\packages\\toolchain-xtensa\\lib\\gcc\\xtensa-lx106-elf\\4.8.2\\include', u'C:\\.platformio\\packages\\toolchain-xtensa\\lib\\gcc\\xtensa-lx106-elf\\4.8.2\\include-fixed', u'C:\\.platformio\\packages\\tool-unity'], 'framework': 'arduino', 'env_path': 'C:\\\\Windows\\\\system32;C:\\\\Windows;C:\\\\Windows\\\\System32\\\\Wbem;C:\\\\Windows\\\\System32\\\\WindowsPowerShell\\\\v1.0\\\\;C:\\\\Programms\\\\Git\\\\cmd;C:\\\\Python27;C:\\\\Python27\\\\Scripts;C:\\\\Programms\\\\CMake\\\\bin;C:\\\\MinGW\\\\bin;C:\\\\Users\\\\Rebecca K\xe4chele\\\\AppData\\\\Local\\\\Microsoft\\\\WindowsApps;', 'systype': 'windows_amd64', 'project_dir': 'C:\\Software\\vfm', u'defines': [u'ARDUINO=20300', u'LWIP_OPEN_SRC', u'F_CPU=80000000L', u'__ets__', u'ICACHE_FLASH', u'PLATFORMIO=30400', u'ESP8266', u'ARDUINO_ARCH_ESP8266', u'ARDUINO_ESP8266_NODEMCU'], 'env_pathsep': ';', u'cxx_path': u'C:\\.platformio\\packages\\toolchain-xtensa\\bin\\xtensa-lx106-elf-g++.exe', 'platformio_path': 'C:\\\\Python27\\\\Scripts\\\\platformio.exe', u'cc_flags': u'-std=gnu99 -Wpointer-arith -Wno-impl [... cut ...]
UnicodeDecodeError: 'utf8' codec can't decode byte 0xe4 in position 231: invalid continuation byte

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

```

ivankravets, 2017-11-02:
Could you try the latest 3.5.0b1?

commit d37c6fcd (2017-05-04): Use root directory for PIO Home when path contains non-ascii characters // Resolve #951 Resolve #952
closed 2018-02-13 by ivankravets