platformio/platformio-core #955: Keyboard/HID library not working with PlatformIO.

simonj0, 2017-05-05:
### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`): `PlatformIO, version 3.4.0b1`

### Description of problem
The Keyboard/HID library does not work after compiling & uploading with platformio, although the same sketch file works with the standard Arduino IDE.

#### Steps to Reproduce

1. Compile & run the code below both in the standard Arduino IDE and from PlatformIO.

### Actual Results
The windows key is pressed after running from the Arduino IDE, but not when run from PlatformIO.

### Expected Results
The windows key should be pressed after both compilations, since it's the same code.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:micro]
platform = atmelavr
board = micro
framework = arduino
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>
#include <Keyboard.h>

void setup() {
  Keyboard.begin();
}

void loop() {
  delay(500);

  Keyboard.press(KEY_LEFT_GUI);
  delay(100);
  Keyboard.releaseAll();

  for(;;);
}
```

simonj0, 2017-05-05:
Alright, the issue seems to be on my side. I tried the same with another Arduino of the same model, and that one seems to work. I suppose it was just faulty hardware, not an issue with PIO; Solved.

closed 2017-05-05 by simonj0

Blockguy24, 2019-10-08:
I don't think this is faulty hardware, I have the same problem on PlatformIO version 4.0.3
First, I upload this code to the Arduino using the Arduino IDE:
```c++
#include <Arduino.h>
#include <Keyboard.h>

void setup() {
  pinMode(6, INPUT_PULLUP);
  Keyboard.begin();
}

void loop() {
  if (!digitalRead(6))
  {
    Keyboard.print("a");
  }
}
```
Then I run this command to read the flash back from the Arduino:
`"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude" -C"C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf" -v -patmega32u4 -cavr109 -PCOM17 -b57600 -D -U flash:r:arduino-ide.hex:i`

I repeated this for the PlatformIO IDE and saw that the .hex files produced were completely different.

[Hex files.zip](https://github.com/platformio/platformio-core/files/3701969/Hex.files.zip)

Also, looking in Devices and Printers, the device is recognized differently:
Arduino IDE:
[image]
[image]

PlatformIO IDE:
[image]
[image]

When it is recognized as a bootloader device, it is still able to run the sketch, except HID functionality does not work.

Blockguy24, 2019-10-09:
Sorry about the previous post, I'll make a new issue as this may not be exactly the same issue.