platformio/platformio-core #956: Serial Monitor won't work today...

flydr2, 2017-05-06:
This is what I get:

Please wait while upgrading PlatformIO...

Platform Atmel AVR
--------
Updating atmelavr                        @ 1.5.0          [Up-to-date]
Updating toolchain-atmelavr              @ 1.40902.0      [Up-to-date]
Updating tool-avrdude                    @ 1.60300.1      [Up-to-date]
Uninstalling framework-arduinoavr @ 1.10617.4:  Error: Traceback (most recent call last):
  File "c:\users\marc\.platformio\penv\lib\site-packages\platformio\__main__.py", line 106, in main
    cli(None, None, None)
  File "c:\users\marc\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\marc\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\marc\.platformio\penv\lib\site-packages\click\core.py", line 1024, in invoke
    Command.invoke(self, ctx)
  File "c:\users\marc\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\marc\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\marc\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\marc\.platformio\penv\lib\site-packages\platformio\__main__.py", line 75, in cli
    maintenance.on_platformio_start(ctx, force, caller)
  File "c:\users\marc\.platformio\penv\lib\site-packages\platformio\maintenance.py", line 66, in on_platformio_start
    after_upgrade(ctx)
  File "c:\users\marc\.platformio\penv\lib\site-packages\platformio\maintenance.py", line 173, in after_upgrade
    if u.run(ctx):
  File "c:\users\marc\.platformio\penv\lib\site-packages\platformio\maintenance.py", line 111, in run
    result.append(callback(ctx))
  File "c:\users\marc\.platformio\penv\lib\site-packages\platformio\maintenance.py", line 154, in _update_dev_platforms
    ctx.invoke(cmd_platform_update)
File "c:\users\marc\.platformio\penv\lib\site-packages\platformio\commands\platform.py", line 373, in platform_update
    platform, only_packages=only_packages, only_check=only_check)

  File "c:\users\marc\.platformio\penv\lib\site-packages\platformio\managers\platform.py", line 124, in update
    self.cleanup_packages(p.packages.keys())
  File "c:\users\marc\.platformio\penv\lib\site-packages\platformio\managers\platform.py", line 149, in cleanup_packages
    pm.uninstall(manifest['__pkg_dir'], trigger_event=False)
  File "c:\users\marc\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 670, in uninstall
    util.rmtree_(pkg_dir)
  File "c:\users\marc\.platformio\penv\lib\site-packages\platformio\util.py", line 592, in rmtree_
    return rmtree(path, onerror=_onerror)
  File "C:\Users\marc\.platformio\python27\Lib\shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "C:\Users\marc\.platformio\python27\Lib\shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "C:\Users\marc\.platformio\python27\Lib\shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "C:\Users\marc\.platformio\python27\Lib\shutil.py", line 256, in rmtree
    onerror(os.rmdir, path, sys.exc_info())
  File "c:\users\marc\.platformio\penv\lib\site-packages\platformio\util.py", line 590, in _onerror
    os.remove(name)
WindowsError: [Error 5] Access is denied: 'C:\\Users\\marc\\.platformio\\packages\\framework-arduinoavr\\libraries\\SD\\src'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues
 
============================================================

flydr2, 2017-05-06:
Fixed it by deleting the "framework-arduinoavr" folder.

ivankravets, 2017-05-06:
Something is wrong with permissions :(

Maybe, you have used PIO Core without IDE with other permissions before.

closed 2017-05-06 by ivankravets