platformio/platformio-core #958: No HTTPS for platformio.org

nopdotcom, 2017-05-09:
What kind of issue is this?

https://platformio.org website.

------------------------------------------------------------------

### Configuration

Ubuntu 16.04.2, OS X 10.11.6, and I stopped trying platforms after that. 

**PlatformIO Version** (`platformio --version`):

Not relevant.

### Description of problem

I want to visit https://platformio.org. I expected it to work like http://platformio.org, except with a padlock.

What I got instead is “This certificate is not valid, because it is for pioplus.com.”

#### Steps to Reproduce

`wget` is less sexy than browser padlocks, but it is reproducible nearly everywhere.

```
$ wget -d https://platformio.org
```

### Actual Results

[noise snipped; you can run wget yourself :–)]

```
DEBUG output created by Wget 1.17.1 on linux-gnu.
--2017-05-09 10:23:08--  https://platformio.org/
Resolving platformio.org (platformio.org)... 193.222.52.25
Initiating SSL handshake.
certificate:
  subject: CN=pioplus.com
  issuer:  CN=Let's Encrypt Authority X3,O=Let's Encrypt,C=US
ERROR: no certificate subject alternative name matches
        requested host name ‘platformio.org’.
```
### Expected Results

```
$ wget https://www.nop.com
--2017-05-09 10:19:01--  https://www.nop.com/
Resolving www.nop.com (www.nop.com)... 72.14.177.231
Connecting to www.nop.com (www.nop.com)|72.14.177.231|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2890 (2.8K) [text/html]
Saving to: ‘index.html.1’
```
[...]

### Additional info

I think you just need a better letsencrypt setup. Send me mail at `nop@nop.com` if you’d like me to peer at it.

ivankravets, 2017-05-09:
Currently, platformio.org doesn't work via HTTPS. We will add support for it later.

ivankravets, 2018-02-12:
Done! https://platformio.org/

closed 2018-02-12 by ivankravets