platformio/platformio-core #959: pip install fails on Travis

SRGDamia1, 2017-05-09:
### Configuration

**Operating system**:  TravisCI

**PlatformIO Version** (`platformio --version`): 3.3.0

### Description of problem
Attempt to install pio on Travis fails at Requests:  "NameError: name 'platform_system' is not defined"
https://travis-ci.org/EnviroDIY/ModularSensors/jobs/230465347

#### Steps to Reproduce

1.  Enable TravisCI with "- pip install -U platformio" in install step
2.  Attempt to run Travis
3.  Error!

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info
Builds were passing for me until just a few hours ago.  I'm not sure what changed.

ivankravets, 2017-05-09:
The issue is linked with PIO Core dependency `requests` and was fixed by @Lukasa https://github.com/kennethreitz/requests/commit/800a074b5d7adbbcaf96a5e6b3754c560f6768ff

Please restart Travis.CI job

closed 2017-05-09 by ivankravets

SRGDamia1, 2017-05-09:
Awesome!  Thanks!