platformio/platformio-core #962: Feature: Don't force "login" to PlatformIO...

madscientist42, 2017-05-11:
People use their Atom editor/ide's for more than just PlatformIO.  If I wanted to be logged into your ecosystem, I would've already- and the latest betas _**OBNOXIOUSLY**_ places itself on top of everything , asking that you "login" or skip when it didn't previously do this.   You want to reconsider such actions as they're annoying enough to make people STOP using the FOSS project...  It's not being helpful to do this.

Back that part out, please.

ivankravets, 2017-05-11:
We have resolved this issue in https://github.com/platformio/platformio-atom-ide/pull/456

We had the same request to GUI for PIO Account because not all developers like to play with CLI.
If you don't want to use extra features from PIO Core+, you can disable PIO Auth in the next beta.

madscientist42, 2017-05-11:
Wicked.  I signed up as a result of this feedback.  It was just very annoying- needed to be something that could be muted.  Since it's fixed, I'll close the issue.  :D

madscientist42, 2017-05-11:
Or, rather, let you do it...since it seems that the initiator can't say, "Ok, it's resolved".  :D

ivankravets, 2017-05-18:
New Beta is coming soon

closed 2017-05-18 by ivankravets