platformio/platformio-core #965: PIO doesn

SRGDamia1, 2017-05-16:
What kind of issue is this?

- [ *] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem

#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

ivankravets, 2017-05-16:
Please move issue to https://github.com/platformio/platform-atmelsam/issues

closed 2017-05-16 by ivankravets

SRGDamia1, 2017-05-16:
Sorry!  Done.