platformio/platformio-core #966: Error when runnning platformio run command

brahbassim, 2017-05-17:
Here is the error i get:
TypeError: CScanner() takes no arguments (1 given):
File "/usr/local/lib/python2.7/site-packages/platformio/builder/main.py", line 103:
env = DefaultEnvironment(**DEFAULT_ENV_OPTIONS)
File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/scons-2.5.1/SCons/Defaults.py", line 88:
_default_env = SCons.Environment.Environment(*args, **kw)
File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/scons-2.5.1/SCons/Environment.py", line 982:
apply_tools(self, tools, toolpath)
File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/scons-2.5.1/SCons/Environment.py", line 107:
env.Tool(tool)
File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/scons-2.5.1/SCons/Environment.py", line 1788:
tool = SCons.Tool.Tool(tool, toolpath, **kw)
File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/scons-2.5.1/SCons/Tool/__init__.py", line 104:
module = self._tool_module()
File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/scons-2.5.1/SCons/Tool/__init__.py", line 119:
return imp.load_module(self.name, file, path, desc)
File "/usr/local/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 79:
class LibBuilderBase(object):
File "/usr/local/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 85:
ADVANCED_SCANNER = SCons.Scanner.C.CScanner(advanced=True)

ivankravets, 2017-05-17:
Please run `brew uninstall scons`

brahbassim, 2017-05-17:
I did it but got the same error...

ivankravets, 2017-05-18:
Please remove this folder `/Users/brah/.platformio/packages/tool-scons/` and start build again

brahbassim, 2017-05-19:
Perfect it works now... Thank you very much.

closed 2017-05-19 by brahbassim