platformio/platformio-core #967: PlaftormIO: Unable to get a list of serial ports.

ilovetogetspamed, 2017-05-24:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Development Platform or Board. All issues related to Development Platforms or Embedded Boards
      should be reported to appropriate repository. 
      See full list with repositories and search for "platform-xxx" repository related to your hardware
      https://github.com/platformio?query=platform-

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [X] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: OS X 10.11.6 (15G1510) 

**PlatformIO Version** (`platformio --version`): PlatformIO, version 2.11.0

### Description of problem
Get error message in popup.  "PlaftormIO: Unable to get a list of serial ports."

#### Steps to Reproduce

1. Try to open Serial Monitor
2.
3.

### Actual Results

### Expected Results
Get list of ports.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info
I can access the device from the command line just fine using 'miniterm.py /dev/cu.usbmodem1411  9600'

ilovetogetspamed, 2017-05-26:
After updating to IDE 2.0.0-beta.4 Core 3.4.0b4 the dropdown populates,
[image]
but when I click on the Serial Monitor button I get:
[image]

ivankravets, 2017-05-26:
Open system terminal (not PIO IDE) and remove obsolete PIO Core.

```
sudo pip uninstall platformio

or 

brew uninstall platformio
```

Restart Atom.

closed 2017-05-26 by ivankravets