platformio/platformio-core #968: internet_on isn't a great test for connectivity

seaduck13, 2017-05-25:
- [X] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem

When behind a firewall, like most professional institutions, it is difficult to get an internet connection for the platformio upgrade/updates.  Checking 8.8.8.8 isn't a great way to see if there is a connection, especially when certain ports are closed for security reasons. 

This is found in util.py

ivankravets, 2017-05-25:
Your proposition?

seaduck13, 2017-05-25:
Why not check for connections to the site/s that are queried for the package updates?  This seems like it'd be more direct and check to make sure that the updates could actually occur, especially since your internet could be on and some other resource could be down.

Alternatively, you could check for a local DNS response for the repo/hosting sites.  That'd be similar to the port 53 connection to 8.8.8.8, correct?

commit 074e262e (2017-05-26): Use BinTray when checking Internet connection // Resolve #968
closed 2017-05-26 by ivankravets