platformio/platformio-core #969: [Feature Request] - Support parsing Arduino additional board json files?

fake-name, 2017-05-28:
I recently purchased a [Arducam ESP8266 module](http://www.arducam.com/world-smallest-esp8266-wifi-camera/), and it only has support for the Arduino Text-editor+macro button build system "IDE".  It also appears to not exist in the platformio framework (not surprising, it's new and kind of esoteric.

Anyways, I was wondering if it would be possible for platformio to parse either the arduino compatible ["index.json"](http://www.arducam.com/downloads/ESP8266_UNO/package_ArduCAM_index.json), or the various metadata files in the underlying [github repository](https://github.com/ArduCAM/ArduCAM_ESP8266_UNO). It *seems* like it should be possible to infer the required configuration from these files to build a binary for the board, but that could be me being naive.

Belisarith, 2017-06-23:
I'll second that request.
The arduino ide json file is located here:
[http://www.arducam.com/downloads/ESP8266_UNO/package_ArduCAM_index.json](url)

You can get working a nodemcu or the arducam uno board working if you select the nodemcu in platform.io and put the arducam library into your lib folder. However, if you do it this way, there is data loss when taking pictures. 
The problem somehow has to do with the standard libraries for the esp8266, see this github link.
[https://github.com/ArduCAM/Arduino/issues/101](url)

If you follow the way described by arducam using the arduino ide the pictures are fine.
So I suppose the makers of arducam somehow modified the esp8266 libraries. But I don't see a possibility to include these modified files when working with platform.io, because always the standard libraries of the framework are compiled. This is way over my level of expertise. Therefore, it would be nice if arducam would be included as one board into platform.io or someone could explain how to force platform.io to take the libraries of the lib folder instead of the framework libraries.

Furthermore, I think it would be huge benefit if there would be some kind of tool to migrate boards by using the json file of the arduino ide and make the necessary migrations to include these boards thereby into platformio. Again way over my level of expertise.

ivykoko1, 2023-01-16:
As the use of generic boards with rare chips and community-made boards becomes more prevalent, it is important to have support for them in the PlatformIO ecosystem. I propose that a "Community Custom Boards and Platforms" repository be created within PlatformIO. This repository would allow users to share information and configurations for custom boards and platforms, making it easier for others to use them in their projects.

_I'll explain two features that I would consider very useful, and should be prioritized as such._

**Arduino to Platformio conversion**

I also propose the development of an automated tool for converting existing Platforms/Boards/Frameworks to PlatformIO. This tool would analyze the existing platform, generate necessary files and configurations, and test for compatibility to ease migration to PlatformIO. This feature would simplify migration and make it more accessible to a wider range of users, as well as reducing the effort of maintaining and updating for multiple platforms
For example, if a user wanted to use a generic board with a rare chip in their project, they could search the repository for that board and find information about its specifications, compatibility with different software and libraries, and any known issues or workarounds. They could also download configuration files and examples directly from the repository, making it easy to get started with the board.

**Community made packages/boards/platforms**

Additionally, by sharing information about custom components in a central location, it would encourage collaboration and knowledge-sharing within the community. This would benefit everyone by making it easier to use custom components and encouraging innovation in the field.

This feature would be beneficial for the community of PlatformIO users, **making it easier to use custom components and encouraging innovation in the field**. It would also help for those that are new to the field to quickly understand the differences between different boards and make a decision on which one to use for their projects.