platformio/platformio-core #973: Missing simba.h header

anweiss, 2017-06-01:
What kind of issue is this?

- [X] PlatformIO Core

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem

When initializing a project using a platform with the Simba framework, it seems to download the SimbaFramework source to `.pioenvs`, but none of the header files are present. Which leads to missing simba.h library errors in my IDE. Builds work fine.

I initially though it was specific to VS Code (https://github.com/platformio/platformio-vscode-ide/issues/8), however, it happens when I import in to CLion as well.

#### Steps to Reproduce

1. Initialize project
2. Open IDE
3. Include `simba.h` header

### Actual Results

Missing simba.h errors

### Expected Results

Headers are present.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:due]
platform = atmelsam
board = due
framework = simba
```

eerimoq, 2017-07-01:
Hello,

Simba version 15.0.0, released today, should fix this issue. I made [the fix suggested by Ivan](https://community.platformio.org/t/simba-linter-in-ide-does-not-find-simba-h/1679/2), and now ``simba.h`` is found by VS Code.

Let me know if there are more fixes/improvements to be implemented to make Simba easier to use using PlatformIO.

Enjoy! =)

ivankravets, 2017-07-01:
@eerimoq 

yes, it works now! :) Thank you!

closed 2017-07-01 by ivankravets