platformio/platformio-core #977: Error while uploading to target (STM32F407, STM32F103)

Split7fire, 2017-06-05:
While trying to upload using stlink in atom (all updates), following error comes out: 

> st-flash: error while loading shared libraries: libstlink-shared.so.1: cannot open shared object file: No such file or directory
> *** [upload] Error 127

------------------------------------------------------------------

### Configuration

**Operating system**: Ubuntu x86_64 17.04

**PlatformIO Version** (`platformio --version`): latest

ivankravets, 2017-06-05:
Please move this issue to https://github.com/platformio/platform-ststm32/issues

closed 2017-06-05 by ivankravets

1am, 2017-08-18:
@ivankravets Where was this issue moved? I can't find it in the repo you mentioned and the issue is still there as I'm experiencing it now.

ivankravets, 2017-08-18:
Please open new here https://github.com/platformio/platform-ststm32/issues

1am, 2017-08-18:
You marked is as moved, thus I'm asking where. Don't want to report issues which are already there.

ivankravets, 2017-08-18:
There is no problem with PlatformIO Core

ivankravets, 2017-08-18:
Duplicate of https://github.com/platformio/platform-ststm32/issues/58