platformio/platformio-core #978: [Feature Request] `upload_resetport` option or better reset autodetection

cinderblock, 2017-06-06:
I recently had issues with PlatformIO detecting which port to use for device reset/uploading new program on an Arduino based Pro Micro. Adding an option similar to `upload_port` for which port to use to send a RTS pulse to reset the target, I believe, would have fixed my problem. Alternatively, an option for `upload_resetmethod` that applies to the Arduino platform could be used to fix this issue. It's also possible that `upload_port` simply needs a little work.

Specifically, I managed to load a program onto my target that resets the target after less than 0.5s. So, the AVR would enumerate as Arduino Bootloader for a few seconds, decide there was no new program, run the main program, which enumerated as a different COM port and then quickly reset.

Therefore, when PlatformIO was told to upload, it most often autodetected the COM port that the Arduino Bootloader had been assigned. This reset did not actually make the AVR reset, and so the COM port never went away. PlatformIO was sitting and waiting for a "new" COM port to send the new program to and never realized it was already talking to the right port.

Setting `upload_port` to the bootloader's fixed COM port did not fix this issue. I was able to get out of this loop by trying to upload a fixed main program repeatedly until the main program's COM port was detected and then the upload process worked as expected.

ivankravets, 2017-06-13:
It is already implemented. See https://github.com/platformio/platformio-core/blob/develop/platformio/builder/tools/pioupload.py#L54

ivankravets, 2017-06-24:
Fixed in https://github.com/platformio/platformio-core/commit/93db0fa06467d514a832992a9bc3e2afaec3b959

Please re-test with http://docs.platformio.org/en/latest/installation.html#development-version

closed 2017-06-24 by ivankravets