platformio/platformio-core #980: Add GDB debugger support for native platform

buzzdavidson, 2017-06-09:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Development Platform or Board. All issues related to Development Platforms or Embedded Boards
      should be reported to appropriate repository. 
      See full list with repositories and search for "platform-xxx" repository related to your hardware
      https://github.com/platformio?query=platform-

- [x] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Mac OSX 10.11.16, Ubuntu Linux 16.04 LTS

**PlatformIO Version** (`platformio --version`): 3.4.0b12

### Description of problem

Some of my projects include shared components targeting desktop machines as well as microcontrollers.  It would be extremely helpful to have the PIO debugger integrated with GDB to enable local native debugging.

#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

qwerty123443, 2020-04-10:
Since there is no progress on this for (close to) 3 years, I'll provide my hotfix for linux (might work on mac as well, haven't tested that):
(note: this breaks debugging for platforms other than native. instructions to undo can be found at the bottom)

1. open  `.vscode/launch.json`
2. delete its contents and replace it with (for example) the following:
```
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "(gdb) Launch",
            "type": "cppdbg",
            "request": "launch",
            "program": "${workspaceFolder}/.pio/build/debug/program",
            "args": [],
            "stopAtEntry": false,
            "cwd": "${workspaceFolder}",
            "environment": [],
            "externalConsole": true,
            "MIMode": "gdb",
            "setupCommands": [
                {
                    "description": "Enable pretty-printing for gdb",
                    "text": "-enable-pretty-printing",
                    "ignoreFailures": true
                }
            ]
        }
    ]
}
```
3. change the file permissions to 444

If you want to change the file, change its permissions back to (f.e.) 666, change the contents and change the permissions back to 444.

BlueAndi, 2020-04-16:
Does that means currently I can not use the windows_x86 platform and debug via mingw gdb?

qwerty123443, 2020-04-16:
To be honest, I haven't thought of windows. You can try generating a configuration automatically from within vscode (there should be a button when you open the `launch.json` file) but I don't know how one would prevent platformio from overwriting it

BlueAndi, 2020-04-16:
I tried this
```
        {
            "name": "(gdb) Starten",
            "type": "cppdbg",
            "request": "launch",
            "program": "${workspaceFolder}/.pio/build/windows_x86/program.exe",
            "args": [],
            "stopAtEntry": false,
            "cwd": "${workspaceFolder}",
            "environment": [],
            "externalConsole": false,
            "MIMode": "gdb",
            "miDebuggerPath": "C:/.platformio/packages/toolchain-gccmingw32/bin",
            "setupCommands": [
                {
                    "description": "Automatische Strukturierung und Einrückung für \"gdb\" aktivieren",
                    "text": "-enable-pretty-printing",
                    "ignoreFailures": true
                }
            ]
        }
```
But my miDebuggerPath seems to bad according to the pop up message.
But this path exsits. Hmm ...

qwerty123443, 2020-04-16:
worst-case you can always use gdb directly (optionally with a custom .gdbinit such as pwndbg).

BlueAndi, 2020-04-16:
Yes, that would really be the worst-case. :-) Having a nice looking IDE and using gdb on console. *LOL*

Schrolli91, 2020-04-27:
@ivankravets hey there - is there any chance to get this feature in the next time?

Atm i'm playing around with jumper.io 's virtual Lab.
But for debugging the code in the sim i need a native GDB solution - see: https://docs.jumper.io/docs/intgdb.html#visual-studio-code

ivankravets, 2020-04-27:
Sorry, currently we do not have plans to implement this. It is still in our TODO list :(

Schrolli91, 2020-04-28:
Is there another way to configure this manually and prevent PIO for overwritting this every time?

qwerty123443, 2020-04-28:
> you can add your config also to the template file:
> https://github.com/platformio/platformio-core/blob/b7ac59066fb9cda8f290ffda32afca020c9a72a5/platformio/ide/tpls/vscode/.vscode/launch.json.tpl
> located under `C:\Users\###\.platformio\penv\Lib\site-packages\platformio\ide\tpls\vscode\.vscode` at my system. Then the auto-update vrom PIO will add your config as well to the given two.

I see, haven't looked at this much. Won't this break with every platformIO update, however? (as, I think, that overwrites your penv folder)

Schrolli91, 2020-04-28:
@qwerty123443  Yes i think you're right - that maybe get lost after an update :(

ivankravets, 2020-04-28:
You can disable it with https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-ide-autorebuildautocompleteindex

Schrolli91, 2020-04-28:
Thanks for this hint - I'll try this.

Is there a way to use a other (local installed) gdb with `debug_tool = custom` and connect to a still running gdbserver over tcp?

gmagno, 2020-05-14:
Hey, great project, congrats!
It would be great to have this feature.

arduhe, 2020-07-07:
To be honest - it's a little bit disappointing that there is no progress in this issue after more than 3 years. This feature would be really helpful.

thinkyhead, 2020-08-15:
I have had success getting the VSCode debugger to launch by pasting the right values into `launch.json`, but of course it doesn't persist….

```json
{
  "name": "Launch Sim",
  "type": "cppdbg",
  "request": "launch",
  "program": "${workspaceFolder}/.pio/build/simulator/TheSimulator",
  "miDebuggerPath": "/usr/local/bin/gdb",
  "MIMode": "gdb",
  "cwd": "${workspaceFolder}/.pio/build/simulator"
}
```

For the native build option in our project we only need support for native `gdb` debugging at the VSCode level and we can launch the VSCode debugger manually. It doesn't need to be integrated with PlatformIO beyond keeping our custom values in `launch.json`. If we could do it just by adding text to `platformio.ini` that would be great, but all we really need is an Advanced Scripting hook that will allow us to modify `launch.json` before it gets written out.

ivankravets, 2020-08-15:
Thanks all for your ideas. Yes, it makes sense to use directly `cppdbg` debugger. We will add support for this soon.

jcw, 2020-09-01:
Not sure what's possible on macOS, where gdb no longer works. I'm always forced to move to Linux for debugging a native build - this is obviously not PIO's fault, but I'd be nice to get _some_ sort of of cmd-line based debugging going - some day.

thinkyhead, 2020-09-10:
> macOS, where gdb no longer works

What's interesting is that I did get the debugger in VSCode working at one point through `ggdb` in the MacPorts install. I don't recall whether I added any extra compile flags or did anything special to get the debugger to start working, but it was showing the current state of variables, single-stepping, and working reliably for a little while there.

However, in my more recent testing I have not been able to get the debugger to run reliably in VSCode, and in fact now it always gets hung up when trying to start the process.

Maybe I just had a brief spot of good weather, but I'll continue to experiment and see whether any reliable debugging is possible on this system with some version of `g++` and `ggdb`.

javiercuadrado, 2020-09-15:
I have this config working on VSCode on Mac. It's generated by default using "add configuration button" and selecting "(lldb) Launch" option.

>        {
>            "name": "(lldb) Launch",
>            "type": "cppdbg",
>            "request": "launch",
>            "program": "${workspaceFolder}/.pio/build/native/program",
>            "args": [],
>            "stopAtEntry": false,
>            "cwd": "${workspaceFolder}",
>            "environment": [],
>            "externalConsole": false,
>            "MIMode": "lldb"
>        }

dillon1110, 2020-09-17:
I am using command palette (Ctrl-Shift-P) and select 'C/C++: Build and Debug Active File'  -> 'gcc build and debug active file' to start debugger. Works on Win10 and remote WSL:Ubuntu environment.

thinkyhead, 2020-09-24:
> It's generated by default using "add configuration button" and selecting "(lldb) Launch" option.

If you make any edit to the `platformio.ini` file the launch option you added will be removed, and you will need to regenerate it again the next time you want to start a debug session. If your binary isn't named `program` (as is typical) then you will need to change the `program` field to match your binary's actual name.

---

As a side topic, I would like to hear from developers on macOS who have gotten either the `gdb` or `gdb-apple` packages (from MacPorts) working reliably. Our native build (and PlatformIO in-general) seems to incompatible with LLVM and requires the use of a `gcc` that has been installed separately, and not all `gcc` / `gdb` combinations are created equal.

I went through the system rigamarole to give `gdb` the permissions it needs to launch and access other programs, and I had a brief success debugging with some combination of gcc / gdb versions. However, since my initial success I have not found a combination of compiler and debugger that can launch without hanging the process.

I'll be sure to ask around the PlatformIO community forum next time I'm on that task.

poelstra, 2020-09-30:
Here's a tip to automatically build before starting a session, and to get working stack traces.

First, put the following in `platformio.ini` (unimportant lines omitted):

```ini
[env:simulator]
platform = native
build_type = debug
```

Then, add the following section to your `launch.json`:

```json
    {
      "name": "Debug simulator",
      "type": "cppdbg",
      "request": "launch",
      "program": "${workspaceFolder}/.pio/build/simulator/program",
      "args": [],
      "environment": [],
      "cwd": "${workspaceFolder}",
      "preLaunchTask": "PlatformIO: Build (simulator)"
    }
```

Note the `preLaunchTask` which refers to build the `simulator` env (as I called it), and the `build_type = debug` in that target.

There is no need to clear out any other debug configurations, just add yours to the existing ones, but as others noted, you'll have to do this every time you make a change to `platformio.ini`.

ivankravets, 2021-03-19:
Debugging for native (desktop) applications is implemented. Please note that there are some issues with our VSCode extension. We will fix them soon. See https://github.com/platformio/platformio-vscode-ide/issues/2447

In any case, using this configuration as @thinkyhead  proposed is a good idea:

```
    {
      "name": "Debug simulator",
      "type": "cppdbg",
      "request": "launch",
      "program": "${workspaceFolder}/.pio/build/simulator/program",
      "args": [],
      "environment": [],
      "cwd": "${workspaceFolder}",
      "preLaunchTask": "PlatformIO: Build (simulator)"
    }
```

The latest PlatformIO Core does not overwrite `launch.json` anymore.

closed 2021-03-19 by ivankravets

ivankravets, 2021-03-19:
A demo

```
pio debug --interface=gdb -x .pioinit
```
[image]

jcw, 2021-04-10:
To follow up on an old comment: native debugging on macOS with gdb still doesn't work for me, but `lldb` works fine and appears to be a drop-in replacement. So debugging on macOS is again possible. Thee's a popup eqch time you start debugging an app, the get permission to place the app under debugger control, which needs authorisation, that's all.

ivankravets, 2021-04-10:
>  native debugging on macOS with gdb still doesn't work for me

I'm on the macOS. The screen above is mine. I used GDB from `brew install gdb`. You also need to generate a certificate, etc. There is instruction for this.

thinkyhead, 2021-04-15:
> The latest PlatformIO Core does not overwrite `launch.json` anymore.

Great! I've been doing a lot of _remote_ debugging lately in the CLI and it'll be great to get back into proper GUI mode.

ivankravets, 2021-04-15:
@thinkyhead please note that the latest PIO Core also has support for native debugging :) See https://github.com/platformio/platformio-core/blob/develop/HISTORY.rst

P.S: Happy to see how PlatformIO helps other open-source projects. Marlin is a great example!!! Great job! 🚀

pharapeti, 2023-11-01:
> I have this config working on VSCode on Mac. It's generated by default using "add configuration button" and selecting "(lldb) Launch" option.

Magic - just tried this and it worked first time with zero additional configuration!

commit 887d4672 (2021-03-19): Debug native (desktop) application on a host machine // Resolve #980