platformio/platformio-core #981: FeatureRequest: Add timestamp to 'platformio device monitor' output

neilyoung, 2017-06-11:
### Configuration

**Operating system**: OSX Sierra, 10.12.5

**PlatformIO Version** (`platformio --version`): PlatformIO, version 3.3.0

### Description of problem

Would it be possible to have an option, which adds a timestamp to the output of 'platformio device monitor'? I have tried various filters which would do that to no avail.
It always end in an exception like this:

```
Exception in thread rx:
Traceback (most recent call last):
  File "/usr/local/Cellar/python/2.7.12_2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/local/Cellar/python/2.7.12_2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/Users/decades/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/serial/tools/miniterm.py", line 453, in reader
    self.console.write(text)
  File "/Users/decades/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/serial/tools/miniterm.py", line 68, in write
    self.output.write(text)
UnicodeEncodeError: 'ascii' codec can't encode character u'\u2418' in position 0: ordinal not in range(128)

```

Reason: Traces from embedded devices, like ESP 8266, most the time do not have a real time reference, just an internal timer or timestamp. Especially if such a device re-boots the absolute time reference is lost and cannot be retrieved easily.

An option, which decorates the line with a preceding timestamp would be great

Example of an exception on using a filter:

```
platformio device monitor --baud 115200  >  ./predate.sh

Sun Jun 11 20:06:53 CEST 2017 : --- Miniterm on /dev/cu.usbserial-FTGXQ6S9 115200,8,N,1 ---
Sun Jun 11 20:06:53 CEST 2017 : --- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H --
-
Sun Jun 11 20:06:57 CEST 2017 : rException in thread rx:
Sun Jun 11 20:06:57 CEST 2017 : Traceback (most recent call last):
Sun Jun 11 20:06:57 CEST 2017 : File "/usr/local/Cellar/python/2.7.12_2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 801, in __bootstrap_inner
Sun Jun 11 20:06:57 CEST 2017 : self.run()
Sun Jun 11 20:06:57 CEST 2017 : File "/usr/local/Cellar/python/2.7.12_2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 754, in run
Sun Jun 11 20:06:57 CEST 2017 : self.__target(*self.__args, **self.__kwargs)
Sun Jun 11 20:06:57 CEST 2017 : File "/Users/decades/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/serial/tools/miniterm.py", line 453, in reader
Sun Jun 11 20:06:57 CEST 2017 : self.console.write(text)
Sun Jun 11 20:06:57 CEST 2017 : File "/Users/decades/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/serial/tools/miniterm.py", line 68, in write
Sun Jun 11 20:06:57 CEST 2017 : self.output.write(text)
Sun Jun 11 20:06:57 CEST 2017 : UnicodeEncodeError: 'ascii' codec can't encode character u'u2418' in position 0: ordinal not in range(128)
Sun Jun 11 20:06:57 CEST 2017 :
```

just with original output instead of an exception.

./predate.sh:

```
#!/bin/bash
while read line ; do
    echo $(date) ":" $line; 
done
```

2ni, 2019-05-08:
you could use timestamp function of http://joeyh.name/code/moreutils/, then do:
```
pio device monitor | ./ts [%H.%M.%.S]
```

ivankravets, 2019-09-04:
We use PySerial. Please forward issue to https://github.com/pyserial/pyserial

closed 2020-03-17 by ivankravets

ivankravets, 2020-03-17:
Please upgrade to the latest version via `pio upgrade --dev` and re-test with `time` filter. See docs https://docs.platformio.org/en/latest/core/userguide/device/cmd_monitor.html

Does it work?

AubsUK, 2022-01-16:
Old, but for anyone else who looks for this.

I didn't need to to install the dev version, so it's been added in to the latest build.

Just open a command prompt in PIO, for me, I don't have PIO path in my Path, so I had to use the fully qualified path `C:\Users\<me>\.platformio\penv\Scripts\platformio.exe`:

```
D:\>C:\Users\<me>\.platformio\penv\Scripts\platformio.exe device monitor --filter time
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Miniterm on COM6  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
18:49:15.708 > ets Jun  8 2016 00:22:57
18:49:15.709 >
18:49:15.709 > rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
```

commit 0df72411 (2020-03-17): Device Monitor Filter API, implement  "time" and "log2file" filters // Resolve #981 Resolve #670