platformio/platformio-core #982: `monitor_baud` option not recognized in platformio.ini

GilDev, 2017-06-12:
### Configuration

**Operating system**: macOS Sierra 10.12.4

**PlatformIO Version**: 3.3.1

Here's a snippet of my platformio.ini:
```
[env:nanoatmega328]
platform = atmelavr
board = nanoatmega328
framework = arduino
monitor_baud = 115200
```

The only thing I changed is this `monitor_baud = 115200` thing. Now, when I `pio run`, I always get this at the top of the command report: `Detected non-PlatformIO `monitor_baud` option in `[env:]` section`

What is happening?

ivankravets, 2017-06-13:
This feature has not been released it. You need http://docs.platformio.org/en/latest/installation.html#development-version

jcollie, 2017-06-28:
This is still happening on version 3.4.0:

```
[jeff@home02 musicbox]$ pio --version
PlatformIO, version 3.4.0
[jeff@home02 musicbox]$ platformio run -t monitor
[Wed Jun 28 18:12:38 2017] Processing huzzah (monitor_baud: 115200; platform: espressif8266; board: huzzah; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Detected non-PlatformIO `monitor_baud` option in `[env:huzzah]` section
Verbose mode can be enabled via `-v, --verbose` option
Collected 26 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <SD(esp8266)> v1.0.5
|   |-- <SPI> v1.0
|-- <ESP8266WiFi> v1.0
|-- <PubSubClient> v2.6
|-- <SPI> v1.0
|-- <IRremoteESP8266> v2.0.3
|-- <Adafruit VS1053 Library> v1.0.3
|   |-- <SD(esp8266)> v1.0.5
|   |   |-- <SPI> v1.0
|   |-- <SPI> v1.0
Linking .pioenvs/huzzah/firmware.elf
Calculating size .pioenvs/huzzah/firmware.elf
text	   data	    bss	    dec	    hex	filename
297934	   9260	  31256	 338450	  52a12	.pioenvs/huzzah/firmware.elf
Building .pioenvs/huzzah/firmware.bin
========================================================================== [SUCCESS] Took 2.41 seconds ==========================================================================
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
```

closed 2017-06-29 by ivankravets

Teiby1, 2018-04-02:
What does homebrew mean? Not official? Because it's still not working with version 3.5.2.

ivankravets, 2018-04-03:
> Because it's still not working with version 3.5.2.

How did you test that?

commit 7a56ec61 (2017-06-29): Added monitor_* options to white-list for Project Configuration File “platformio.ini” // Resolve #982