platformio/platformio-core #984: Uncaught Error: Failed to get boards

54696d21, 2017-06-21:
Atom: 1.18.0 x64
Electron: 1.3.15
OS: DESKTOP-JLMUDUI
Thrown From: linter-ui-default package 1.6.2

Stack Trace

Uncaught Error: Failed to get boards

At C:\Users\gabda\.atom\packages\linter-ui-default\node_modules\react-dom\lib\ReactErrorUtils.js:51

Error: Failed to get boards
    at getBoards (/packages/platformio-ide/lib/utils.js:279:13)
    at InitializeNewProjectView.initializeBoardsSelect (/packages/platformio-ide/lib/init/view.js:50:20)
    at InitializeNewProjectView.initialize (/packages/platformio-ide/lib/init/view.js:46:10)
    at InitializeNewProjectView.BaseView (/packages/platformio-ide/lib/base-view.js:16:20)
    at /packages/platformio-ide/lib/init/view.js:16:38)
    at Object.command (/packages/platformio-ide/lib/init/command.js:20:14)
    at /packages/platformio-ide/lib/main.js:100:59)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (~/AppData/Local/atom/app-1.18.0/resources/app/src/command-registry.js:265:35)
    at CommandRegistry.handleCommandEvent (~/AppData/Local/atom/app-1.18.0/resources/app/src/command-registry.js:3:65)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (~/AppData/Local/atom/app-1.18.0/resources/app/src/command-registry.js:166:25)
    at Object.runAtomCommand (/packages/platformio-ide/lib/utils.js:252:24)
    at callback (/packages/platformio-ide/lib/home/containers/welcome-page.js:26:31)
    at Object.invokeGuardedCallback (/packages/linter-ui-default/node_modules/react-dom/lib/ReactErrorUtils.js:26:5)
    at executeDispatch (/packages/linter-ui-default/node_modules/react-dom/lib/EventPluginUtils.js:85:21)
    at Object.executeDispatchesInOrder (/packages/linter-ui-default/node_modules/react-dom/lib/EventPluginUtils.js:108:5)
    at executeDispatchesAndRelease (/packages/linter-ui-default/node_modules/react-dom/lib/EventPluginHub.js:43:22)
    at executeDispatchesAndReleaseTopLevel (/packages/linter-ui-default/node_modules/react-dom/lib/EventPluginHub.js:54:10)
    at Array.forEach (native)
    at forEachAccumulated (/packages/linter-ui-default/node_modules/react-dom/lib/forEachAccumulated.js:24:9)
    at Object.processEventQueue (/packages/linter-ui-default/node_modules/react-dom/lib/EventPluginHub.js:254:7)
    at runEventQueueInBatch (/packages/linter-ui-default/node_modules/react-dom/lib/ReactEventEmitterMixin.js:17:18)
    at /packages/linter-ui-default/node_modules/react-dom/lib/ReactEventEmitterMixin.js:27:5)
    at handleTopLevelImpl (/packages/linter-ui-default/node_modules/react-dom/lib/ReactEventListener.js:72:24)
    at ReactDefaultBatchingStrategyTransaction.perform (/packages/linter-ui-default/node_modules/react-dom/lib/Transaction.js:143:20)
    at Object.batchedUpdates (/packages/linter-ui-default/node_modules/react-dom/lib/ReactDefaultBatchingStrategy.js:62:26)
    at Object.batchedUpdates (/packages/linter-ui-default/node_modules/react-dom/lib/ReactUpdates.js:97:27)
    at dispatchEvent (/packages/linter-ui-default/node_modules/react-dom/lib/ReactEventListener.js:147:20)

-0:17.9.0 tree-view:show (atom-workspace.workspace.scrollbars-visible-always)
     -0:08.2.0 platformio-ide:initialize-new-project (atom-workspace.workspace.scrollbars-visible-always.theme-one-light-syntax.theme-one-light-ui)

atom-beautify 0.30.2 
atom-material-ui 2.0.4 
auto-indent 0.5.0 
autocomplete-clang 0.11.3 
build 0.68.0 
busy 0.7.0 
docblockr 0.10.3 
file-icons 2.1.7 
intentions 1.1.2 
language-ini 1.19.0 
linter 2.2.0 
linter-gcc 0.7.1 
linter-ui-default 1.6.2 
material-ui 1.0.10 
minimap 4.28.2 
platformio-aia 0.0.5 
platformio-ide 2.0.0-beta.6 
platformio-ide-debugger 1.2.2 
platformio-ide-terminal 2.5.1 
tool-bar 1.1.0

ivankravets, 2017-06-21:
Please move issue to https://github.com/platformio/platformio-atom-ide/issues

ivankravets, 2017-06-21:
http://docs.platformio.org/en/latest/faq.html#multiple-pio-cores-in-a-system

Also, try re-install PIO IDE and remove PIO Core http://docs.platformio.org/en/latest/ide/atom.html#uninstall-atom-with-platformio-ide

closed 2017-06-21 by ivankravets