platformio/platformio-core #986: pio device monitor cannot list all COM ports if their description contains non-ASCII characters

AndreKR, 2017-06-23:
What kind of issue is this?

- [X] PlatformIO Core. If you’ve found a bug, please provide an information below.

### Configuration

**Operating system**: Windows 7 x64 German
**PlatformIO Version**: 3.4.0b12

### Description of problem

```
> pio device monitor

--- Available ports:
---  1: COM6                 Intel(R) Active Management Technology - SOL (COM6)
---  2: COM13                F5521gw Mobile Broadband Device Management (COM13)
---  3: COM14                F5521gw Mobile Broadband GPS Port (COM14)
Error: 'ascii' codec can't encode characters in position 9-10: ordinal not in range(128)
```

The next COM port that would follow is called `Standardmäßgige Seriell-über-Bluetooth-Verbindung`, so I guess the `ä` and `ß` is to blame here.

ivankravets, 2017-06-24:
Does `pio device list` command work?

AndreKR, 2017-06-24:
Yes.

ivankravets, 2017-06-24:
Could you try
```
python -m serial.tools.miniterm
```

Does it fail? If yes, please move this issue to https://github.com/pyserial/pyserial/issues

ivankravets, 2017-06-24:
To avoid this issue you can use:

* http://docs.platformio.org/en/latest/projectconf/section_env_monitor.html#projectconf-monitor-port
* http://docs.platformio.org/en/latest/userguide/cmd_device.html#cmdoption-platformio-device-monitor-p

AndreKR, 2017-06-24:
Yep, fails. I'll open an issue with them.

`pio device monitor`:
```
--- Available ports:
---  1: COM6                 Intel(R) Active Management Technology - SOL (COM6)
---  2: COM13                F5521gw Mobile Broadband Device Management (COM13)
---  3: COM14                F5521gw Mobile Broadband GPS Port (COM14)
Error: 'ascii' codec can't encode characters in position 9-10: ordinal not in range(128)
```

`pio device list`:
```
COM13
-----
Hardware ID: USB VID:PID=0BDB:1911 LOCATION=1-1.4.4
Description: F5521gw Mobile Broadband Device Management (COM13)

COM16
-----
Hardware ID: USB VID:PID=1B4F:9205 SER=7 LOCATION=1-1.5.5
Description: SparkFun Pro Micro bootloader (COM16)

COM6
----
Hardware ID: PCI\VEN_8086&DEV_1C3D&SUBSYS_21CF17AA&REV_04\3&21436425&0&B3
Description: Intel(R) Active Management Technology - SOL (COM6)

COM14
-----
Hardware ID: ERICSSON\GPS\B113C87391F2FED0_09
Description: F5521gw Mobile Broadband GPS Port (COM14)

COM15
-----
Hardware ID: BTHENUM\{00001101-0000-1000-8000-00805F9B34FB}_LOCALMFG&0000\8&2D4DBFD6&0&000000000000_00000004
Description: Standardmäßgige Seriell-über-Bluetooth-Verbindung (COM15)
```

`python -m serial.tools.miniterm`:
```

--- Available ports:
---  1: COM6                 Intel(R) Active Management Technology - SOL (COM6)
---  2: COM13                F5521gw Mobile Broadband Device Management (COM13)
---  3: COM14                F5521gw Mobile Broadband GPS Port (COM14)
Traceback (most recent call last):
  File "C:\Users\andre\.platformio\python27\Lib\runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "C:\Users\andre\.platformio\python27\Lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "C:\Users\andre\.platformio\penv\lib\site-packages\serial\tools\miniterm.py", line 930, in <module>
    main()
  File "C:\Users\andre\.platformio\penv\lib\site-packages\serial\tools\miniterm.py", line 857, in main
    args.port = ask_for_port()
  File "C:\Users\andre\.platformio\penv\lib\site-packages\serial\tools\miniterm.py", line 325, in ask_for_port
    sys.stderr.write('--- {:2}: {:20} {}\n'.format(n, port, desc))
UnicodeEncodeError: 'ascii' codec can't encode characters in position 9-10: ordinal not in range(128)
```

closed 2017-06-24 by AndreKR