platformio/platformio-core #989: Error extracting downloaded frameworks

matteofumagalli1275, 2017-06-26:
### Configuration

**Operating system**: Windows

**PlatformIO Version** (`platformio --version`): 3.3.1

### Description of problem
When platformio is fetching libraries from the framewrok's repositories it fails extracting the archive.

For example it happened with *framework-mbed-3.142.0.tar.gz*. It downloads the archive as a temporary directory  but then the process prompts the error:

`Could not install 'framework-mbed' with version requirements '~3.142.0' for your system 'windows_x86'.`

If i move the archive out from the temporary directory and manually extract it, the process continues and i can build the project successfully.

I used the same project many time last week without issues, platformio correctly downloded all the dependencies without problems.

#### Steps to Reproduce

1. `git clone https://github.com/rushup/Cloud-JAM-Platformio`
2. `platformio init`
3. `platformio update`
4. `platformio run`

### Actual Results
`Could not install 'framework-mbed' with version requirements '~3.142.0' for your system 'windows_x86'.`

### Additional info

I don't have any antivirus running.

Additional log:

```
PackageManager: Installing framework-mbed @ ~3.142.0
Downloading  [####################################]  100%
Unpacking  [###---------------------------------]    9%  00:00:13
Warning! Package Mirror: [Error 32] Impossibile accedere al file. Il file Þ utilizzato da un altro processo: 'C:\\Users\\matteo.fumagalli\\Downloads\\Cloud-JAM-Platformio-master\\.platformio\\packages\\_tmp_installing-0efcpp-package\\framework-mbed-3.142.0.tar.gz'
Looking for other mirror...
```

The error is in italian but basically it says that  it can't open a file.

First thing i thought is a problem with my pc, but i tried another pc starting from a clean enviroment and i get the same error.
As i said before, last week everything worked...

Any idea?

Thanks.

ivankravets, 2017-06-26:
Hm... Strange issue. Could you reproduce this issue after PC restart? Also, would be good if you remove `$HOME/.platformio` folder.

matteofumagalli1275, 2017-06-27:
Hi,

I tried the following:
- Restarting the PC -> Same result
- Deleting .platformio folder -> Same result
- Changing the platformio.ini so that it uses the common path by removing the home_dir field -> Same result

Could you follow steps to reproduce in your pc and tell me if it works for you?

I just wanted to report a possible common issue since a colleague of mine has the same problem. As i said i can skip the error just by extracting the mbed archive manually.

Thanks.

ivankravets, 2017-06-27:
We have thousands of users per day and you are the first with this issue. Could you provide your `platformio.ini`? It seems that your broke `home_dir` option.

matteofumagalli1275, 2017-06-27:
The project i'm using is this:

https://github.com/rushup/Cloud-JAM-Platformio

The ini is the following:
```
[platformio]
home_dir = ./.platformio

[env:nucleo_f401re]
platform = ststm32
board = nucleo_f401re
framework = mbed

upload_port = H:/

monitor_port = COM5

monitor_baud = 9600

    
build_flags = -I./src/X_NUCLEO_NFC01A1 -I./src/X_NUCLEO_NFC01A1/m24sr -I./src/X_NUCLEO_NFC01A1/m24sr/Common
    -I./src/X_NUCLEO_NFC01A1/Common -I./src/X_NUCLEO_NFC01A1/ST_INTERFACES/Communications
    -I./src/NDefLib -I./src/NDefLib/RecordType -I./src/X_NUCLEO_IKS01A2 -I./src/X_NUCLEO_IKS01A2/ST_INTERFACES/Communications 
    -I./src/X_NUCLEO_IKS01A2/ST_INTERFACES/Common -I./src/X_NUCLEO_IKS01A2/ST_INTERFACES/Sensors -I./src/X_NUCLEO_IKS01A2/ST_INTERFACES/Actuators
    -I./src/X_NUCLEO_IKS01A2/Components -I./src/X_NUCLEO_IKS01A2/Components/HTS221Sensor -I./src/X_NUCLEO_IKS01A2/Components/LPS22HBSensor
    -I./src/X_NUCLEO_IKS01A2/Components/LSM6DSLSensor -I./src/X_NUCLEO_IKS01A2/Components/LSM303AGRSensor -I./src/X_NUCLEO_IKS01A2/X_NUCLEO_COMMON/DgbMCU
    -I./src/X_NUCLEO_IKS01A2/X_NUCLEO_COMMON/DevI2C -I./src/X_NUCLEO_IKS01A2/X_NUCLEO_COMMON/DevSPI
    -I./src/X_NUCLEO_IDW01M1v2 -I./src/X_NUCLEO_IDW01M1v2/SPWF01SA -I./src/X_NUCLEO_IDW01M1v2/SPWF01SA/ATParser
    -I./src/X_NUCLEO_IDW01M1v2/SPWF01SA/ATParser/BufferedSerial -I./src/X_NUCLEO_IDW01M1v2/SPWF01SA/ATParser/BufferedSerial/Buffer
    -I./src/NetworkSocketAPI -I./src/NetworkSocketAPI/DnsQuery
    -I./src/HTTPClient -I./src/HTTPClient/data
    -I./src/MQTT -I./src/MQTT/FP -I./src/MQTT/MQTTPacket  

lib_ignore = mbed-rtos, nsapi, lwip, mbed-net
```

Anyway following your suggestion i got some news:

Without the *home_dir* parameter it works. When i tried before i commented the filed using "#" instead of the semicolon :/

Seems like the issue is that parameter, i'm using it to create a standalone project and last week it worked fine.

Do you see something wrong in what i did?

Thanks.

ivankravets, 2017-06-27:
> When i tried before i commented the filed using "#" instead of the semicolon :/

 "#" should work too, be sure that there is no spaces before "#". That is a requirement of INI format.

> Do you see something wrong in what i did?

Why do you need to override PIO Core Home directory? PIO Core installs here platform dependent toolchains, libraries, and other instruments. This folder is not portable between different OS or architectures.

matteofumagalli1275, 2017-06-28:
Hi,

I changed the home directory to create a standalone project, similar to to what VisualGDB does.

Since we were helding an Hackathon with many people it was better to provide an archive with most of the stuff already downloaded. Everyone had the same OS.

Also, when you complete a project shouldn't you be able to backup the full source code? If after years you have to change something, you may face issues by upgraded dependencies or missing online services.

I saw your answer [here](https://community.platformio.org/t/stand-alone-mode/1216) at a similar question and i thought it should have worked with the *home_dir* parameter too.

Anyway, i just thought it was a cool feature to use that way.

ivankravets, 2017-06-28:
If they use the same OS, you can share with them own PIO Core Home. Try this:

1. Add this to `platformio.ini`

```
[platformio]
home_dir = .platformio
```

2. Build a project, upload a firmware, launch debugger. PIO should download all dependencies.

3. Remove `.platformio/appstate.json`

4. Pack whole project. It should work on another machine with the same OS and architecture.

matteofumagalli1275, 2017-06-28:
I still get the error during the download of the dependencies but this time i see an additional exception:

```
Downloading  [####################################]  100%
Unpacking  [###---------------------------------]    9%  00:01:14
Exception AttributeError: "'FileDownloader' object has no attribute '_request'" in <bound method FileDownloader.__del__ of <platformio.downloader.FileDownloader object at 0x0000000004E1E9E8>> ignored
Warning! Package Mirror: [Error 32] Impossibile accedere al file. Il file Þ utilizzato da un altro processo: 'C:\\Users\\matteo.fumagalli\\Downloads\\Cloud-JAM-Platformio-master\\.platformio\\packages\\_tmp_installing-mxptyk-package\\framework-mbed-3.142.0.tar.gz'
Looking for other mirror...
```

It always gets stuck while unpacking the mbed framework.

Later i'll try on another computer and update the result.

matteofumagalli1275, 2017-06-28:
This time i tried the same on my home PC and it worked without issues.
I'm getting crazy.

I'll do some additional tests and let you know.

ivankravets, 2017-06-28:
> It always gets stuck while unpacking the mbed framework.

Does it unpack other packages correctly?

matteofumagalli1275, 2017-06-28:
Ok,
I managed to make it work on the other pc too.

The issue seems just the path that gets too long when extracting the mbed framework.
I was trying the project on the Download directory or in the Desktop. Moving the project to a C:/Test folder makes it work.

I tried also to enable the NTFS's LongPath and rebooting but didn't see any change.

PS: Ignore the 'AttributeError' python exception above, that was because of a different version of Python that i was trying.

matteofumagalli1275, 2017-06-28:
> Does it unpack other packages correctly?

Yes

ivankravets, 2017-06-28:
This is known issue https://support.microsoft.com/en-us/help/830473/command-prompt-cmd.-exe-command-line-string-limitation

matteofumagalli1275, 2017-06-28:
Well i got a similar behaviour even with just a simple python program using tarfile, so i don't think it's just an issue related to the windows's cli.

Anyway i think we can close the issue since it doesn't seem a platformio problem.

Thanks,

ivankravets, 2017-06-28:
To avoid issues with "[platformio] home_dir = .platformio", you can ask users to place ".platformio" folder to %HOME_DIR% folder. In this case, not matter will the user use the same folder of a project it will have pre-installed packages.

For example,

1. Build & Debug project
2. Pack 3 folders "packages, platforms and lib" from "%HOME_DIR%/.platformio"

New user:

1. Navigate to `%HOME_DIR%` and create `.platformio` folder
2. Unpack "packages, platforms and lib" here
3. Launch PlatformIO IDE for Atom or VSCode

------

In any case, is better when PIO automatically install all dependent packages.

closed 2017-06-28 by ivankravets

KamiKazeK, 2017-11-08:
I just discovered that it's not an error with platformio but with the VSCode shell instead. The integrated shell behaves weird sometimes with some commands, cause when you try to manually install the packages in a regular cmd window outside of VSCode it works fine.