platformio/platformio-core #991: Issue with not compiling on platformio. Arduino compiles OK

ShikOfTheRa, 2017-06-29:
Hi. Firstly what an awesome project this is. Been using it for a while as part of travis but just migrating to it running under ubunto.

For Arduino sketch in MW_OSD within this repository - https://github.com/ShikOfTheRa/PIO-SCARAB-TEST 

I have issues with a few of the variants not compiling with platformio. It compiles fine under Arduino - various version tested.

I guess it’s something to do with pre-processor ordering to working in same way as Arduino – or not liking the #defines

Any suggestions or is this a bug?

------------------------------------------------------------------

### Configuration

Ubunto/Atom or Travis

**PlatformIO Version** (`platformio --version`):

core 3.4.1.1a1
ide 2.0.0.0-beta.7 

Operating System Details
Distributor ID:	Ubuntu
Description:	Ubuntu 12.04.5 LTS
Release:	12.04
Codename:	precise
Linux Version
3.13.0-29-generic

#### Steps to Reproduce

1. Compile code in  https://github.com/ShikOfTheRa/PIO-SCARAB-TEST with #define GPSOSD_UBLOX enabled
2. or look at https://travis-ci.org/ShikOfTheRa/PIO-SCARAB-TEST/builds/248294729
3.

### Actual Results
Fails to compile - function not declared in scope (works in Arduino)

### Expected Results
Compile OK

### If problems with PlatformIO Build System:

### Additional info

valeros, 2017-06-29:
Hi @ShikOfTheRa !
I see there is a mix of ino/cpp files in your project and that isn't a great idea.
Instead of "Arduino style" ino files with preprocessing, I propose you to completely switch to the proper project structure with header files (declared interface for the module) and source files (implementation of the module)

ivankravets, 2017-06-29:
http://docs.platformio.org/en/latest/faq.html#convert-arduino-file-to-c-manually

ShikOfTheRa, 2017-06-29:
Hi @valeros yes  agree - in an ideal world I would like to do that. 
It has grown piecemeal over the years through many contributors, styles, merging of other code etc, Many are not programmers. Arduino IDE is so easy and enables them to contribute. Probably why it has been so successful. 
So I have a choice to do that or revert close to Arduino.

ShikOfTheRa, 2017-06-29:
HI @ivankravets thanks also for your response. 
So I guess the answer is for someone to review and move closer to stricter requirements for compiling one way or another.
I don't relish the though of tidying up the entire project just at this moment due to time constraints although I would dearly like to to that. Think I will carry out minimal work such as re-ordering until have the time free.

Thanks again both. i 'll leave open for a couple of days for any other comments and then close out.

ivankravets, 2017-06-29:
The issue is linked with our parser for prototypes. It doesn't handle "whitespaces" before prototype declarations. See [RegExp](https://github.com/platformio/platformio-core/blob/develop/platformio/builder/tools/piomisc.py#L33), which ignore methods from classes and only parses prototypes that start without whitespaces

I've just shifted all code in `GPS.ino` TO LEFT, and your project builds successfully.

How is difficult for you to move a few prototypes to left? Yes, that is really a hook for PlatformIO, but it will allow using project in Arduino IDE and PlatformIO.

ShikOfTheRa, 2017-06-29:
@ivankravets ah I missed the whitespace. That is no issue to amend.

Perfect. I can revisit the whole project at leisure.

Many thanks both for guidance and a resolution.

closed 2017-06-29 by ShikOfTheRa