platformio/platformio-core #994: Problems compiling

daecepi, 2017-07-07:
### Configuration

**Operating system**:
Windows 8.1

**PlatformIO Version** (`platformio --version`): 2.0 beta (and in the version before i had the same issue)

### Description of problem
Problems when trying to compile the code in platformio

#### Steps to Reproduce
1.After getting every part of the code wrtten, i press the compile buttton

### Actual Results
Image of my results:
[image]

### Expected Results
Compiling succesfull or something like that

### If problems with PlatformIO Build System:
[env:atmelavr_arduino_uno_board]
platform = atmelavr
framework = arduino
board = uno
upload_speed = 115200
targets = upload

**Source file to reproduce issue:**
#include <Arduino.h>

int led = 13;

void setup(/* arguments */) {
  //Serial.begin(9600);
  pinMode(led, OUTPUT);
}

void loop() {
  delay(1000);
  digitalWrite(led, HIGH);
  delay(1000);
  digitalWrite(led, LOW);
}

### Additional info

ivankravets, 2017-07-07:
Could you reproduce this issue when open PlatfomIO IDE Terminal and type `pio run`?

daecepi, 2017-07-08:
Yes
[image]
This occurs when running from the command's tool

And when i try to change the board, the problem seems to change
[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
upload_speed = 115200

[image]

ivankravets, 2017-07-09:
Could you try to remove `.platformio` folder from HOME directory and start build again?

ivankravets, 2017-07-09:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
python --version
pip --version
pip list
```

daecepi, 2017-07-10:
There is no .platformio folder in the project's folder, and these are the captures from the commands you requested:
[image]
[image]
[image]
As you see the command "pip" was not recongnized from a normal opened terminal (by the way i tried the command pio as well, and neither does it)

Thank's again

daecepi, 2017-07-10:
1 correction when i run the command "platformio --version" this is the output: PlatformIO, version 3.4.1a5

ivankravets, 2017-07-10:
Could you try to create project in "Documents/Test" folder  (without spaces) and try again. Just for test

daecepi, 2017-07-11:
I tried it as you mentioned and the problem remains, i have a question is necesary to have the board connected for compilation purposes?

[image]

daecepi, 2017-07-11:
And when running from the command line:

[image]

ivankravets, 2017-07-19:
Do you have antivirus in a system? Maybe, McAfee?

daecepi, 2017-07-24:
Problem solved, i installed LLVM software with the version recommended for platformio installing tutorials.

Somehow the version of the software was newer, eventhough i had it installed with the right version.

Everything works fine now, except for the serial monitor which shows an empty list in the ports selection and when trying to upload  shows me the message:
[image]

Any suggestions?
Thanks in advance

ivankravets, 2017-07-24:
Do you see ports in OS Device Manager (COM section)?

daecepi, 2017-07-24:
I forgot to mention i changed the OS recently, now is a windows 10 home single language (because of some problems i had on the last one, aside from that problem all software worked properly by the way)

Now the ports appear to be identified like this:
[image]

always varying last number of the port section:
Port_#0003.Hub_#0006
Port_#0004.Hub_#0006
.
.
.

ivankravets, 2017-07-27:
Could I close this issue?

daecepi, 2017-07-28:
Yes, sure

closed 2017-07-28 by ivankravets