platformio/platformio-core #996: board mightycore644 fails to find SPI.h

zerog2k, 2017-07-07:
fails to compile with
`#include <SPI.h>` on board=mightycore644, but works on similar board, e.g. 644pa8m

maybe @MCUdude is aware of this?

full output:
```
Device: atmega644p

Program:   13158 bytes (20.1% Full)
(.text + .data + .bootloader)

Data:        817 bytes (19.9% Full)
(.data + .bss + .noinit)

================================================= [SUCCESS] Took 8.49 seconds =================================================

========================================================== [SUMMARY] ==========================================================
Environment mightycore644       [ERROR]
Environment 644pa8m             [SUCCESS]
================================================= [ERROR] Took 10.16 seconds =================================================
The terminal process terminated with exit code: 1
```

ivankravets, 2017-07-09:
This is our bug, please move this issue to https://github.com/platformio/platform-atmelavr/issues

closed 2017-07-09 by ivankravets

ivankravets, 2017-07-09:
Fixed in https://github.com/platformio/platformio-pkg-framework-arduinoavr/commit/1228d27325f3f1890b455474fea02b64e3f6e7fe

Please update all PIO packages via `pio update` command or use GUI in PIO Home for Platforms.

zerog2k, 2017-07-09:
@ivankravets thanks!