platformio/platformio-core #997: PlatformIO installation script failed on Ubuntu 16.04 and Raspbian

moisesfr, 2017-07-08:
Hello,

While trying to install using the same methods that always worked, now I get this error on installation.
Have tried on Ubuntu and on a fresh Raspbian OS but I'm getting the same error:

==> Installing Python Package Manager ...
Collecting pip
  Downloading pip-9.0.1-py2.py3-none-any.whl (1.3MB)
Installing collected packages: pip
  Found existing installation: pip 1.5.6
    Uninstalling pip-1.5.6:
      Successfully uninstalled pip-1.5.6
Successfully installed pip-9.0.1

[SUCCESS]

==> Installing PlatformIO and dependencies ...

/usr/bin/python: can't open file 'pip': [Errno 2] No such file or directory

[FAILURE]
The installation process has been FAILED!
Please report about this problem here
< https://github.com/platformio/platformio-core/issues >

ivankravets, 2017-07-09:
Internet connection?

moisesfr, 2017-07-10:
Hi Ivan,

Internet connection apparently with no problems.

I've just re-tried "sudo -H pip install -U platformio" and apparently now it worked. Repeated the process just to check and it worked again but with an error in between: 
Retrying (Retry(total=4, connect=None, read=None, redirect=None)) after connection broken by 'ProtocolError('Connection aborted.', error(104, 'Connection reset by peer'))': /simple/platformio/ 

The install script error however seems to persist: 
sudo python -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/develop/scripts/get-platformio.py)"

==> Installing Python Package Manager ...
Requirement already up-to-date: pip in /usr/local/lib/python2.7/dist-packages

[SUCCESS]

==> Installing PlatformIO and dependencies ...

/usr/bin/python: can't open file 'pip': [Errno 2] No such file or directory

[FAILURE]
The installation process has been FAILED!
Please report about this problem here
< https://github.com/platformio/platformio-core/issues >

Thank you for your attention

ivankravets, 2017-07-10:
If you have `pip` in a system, you can use `sudo -H pip install -U platformio`. It seems that from "root" there is no `pip` in a $PATH.

Nevertheless,  `sudo -H pip install -U platformio` is a right way.

closed 2017-07-10 by ivankravets

moisesfr, 2017-07-10:
Ok, thank you.

I've tried to re-install when I started to get some strange errors compiling sketches, even a sketch with an empty loop and setup method:

 /tmp/ccWQPL86.ltrans0.ltrans.o: In function `main':
ccWQPL86.ltrans0.o:(.text.startup+0x86): undefined reference to `setup'
ccWQPL86.ltrans0.o:(.text.startup+0x8e): undefined reference to `loop'
collect2: error: ld returned 1 exit status
*** [.pioenvs/uno/firmware.elf] Error 1
========================== [ERROR] Took 14.40 seconds ==========================

Don't now what have changed but can't compile.

Should I open an issue addressing this?

ivankravets, 2017-07-10:
> Should I open an issue addressing this?

No, you should not. GCC gave you the hints... `undefined reference to `loop'`

ivankravets, 2017-07-10:
Check that you have `main.cpp` in `src` folder and implemented setup/loop functions.

moisesfr, 2017-07-10:
You're right, it's completely on me, the sketch.ino is not getting there correctly, so no setup or loop or nothing. 

Thanks again.