platformio/platformio-core #999: Error finding serial-ports on Win8 (since last upgrade)

IoTAdri, 2017-07-17:
- [X] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

Win8.1

**PlatformIO Version** (`platformio --version`):

### Description of problem
Cant find list of Serial-ports (see picture)

#### Steps to Reproduce

1. Open Serial-terminal

### Additional info
[image]

IoTAdri, 2017-07-18:
could it be that core 3.4.**1a6** is installed (see picture) and the errormessage states that "3.4.**1a5** distribution is required by the application".. how to correct this? please advise

ivankravets, 2017-07-19:
1. Please opem system Terminal `cmd` (not in IDE), and check that you don't have global PIO Core. Run `pip uninstall platformio`. If you see error that `pip` not found or etc, that is OK.
2. Remove `C:users\adri\.platformio\penv` folder and restart Atom

closed 2017-07-19 by ivankravets